nightscout/AndroidAPS #919: Dana wrong password

ulastojak, 2021-11-13:
Hi 
I made update to version 2.8.2.1 and imported all previous settings from 2.6.2. but i'm getting info that password to pump is not correct and i can't make bolus. I have dana r pump and phone with android 11. My phone is connected to pump so password is correct and when i returned to 2.6.2 everything is working. Any ideas how to connect pump?
[image]
[image]
[image]

MilosKozak, 2021-11-15:
in 2.6.2 password was not verified for RS v1 pump. In 2.8.2 it is
did you read this? https://androidaps.readthedocs.io/en/latest/Configuration/DanaRS-Insulin-Pump.html

closed 2021-11-15 by MilosKozak

ulastojak, 2021-11-15:
I have old Dana Diabecare pump. Not rs version
[image]

MilosKozak, 2021-11-15:
then you  can use code (menu) 3463 to change user pass

ulastojak, 2021-11-16:
There is no option of changing password on my pump. I found old chat with Paul Janiak from 2020 so i guess i have the same issue. Dana r from 2013 and can't change password with code 3463. New version of aps won't work without new firmware?

Paul Janiak @Pallesauka May 13 2020 09:32
My pump was Soo old that I could not change pump default password with the 3463 code. Had to change to a different DanaR with the v2 firmware to be able to change pump password. Now AAPS connects again
https://gitter.im/MilosKozak/AndroidAPS?at=5eb957e13f5b01031940146d

MilosKozak, 2021-11-16:
i don't have so old pump to be able test it but generaly AAPS will work if you find the password
you may ask on FB group if someone knows or upgrade pump to v2

Pallesauka, 2021-11-16:
I can confirm that after a certain AAPS version upgrade we could no longer connect using the default pump pairing password. I guess this was a safety feature that was introduced. It was also not possible to change password with the 3436 code in this old firmware. After firmware upgrade everything worked well again, i.e. It was possible to change pump password using 3436 code and AAPS connect without issues.
I suggest that you update the firmware version on your Dana R.

MilosKozak, 2021-11-16:
one more option is search for password in log files

ulastojak, 2021-11-16:
Hi Paul do you know how to update firmware?

Pallesauka, 2021-11-16:
Ask around in the Facebook group. Some people have access to firmware update tools. Or, if you have a local Dana representative in your country they might assist you.

ulastojak, 2021-11-30:
Hi. I contacted pump producer and they confirmed that it's not possible to change password on my pump. Is this pairing change really necessary in new version of aaps? Local distributor is still selling old pumps so buying new one is not an option, Dana RS is not available.

qq254344379, 2023-06-21:
> I can confirm that after a certain AAPS version upgrade we could no longer connect using the default pump pairing password. I guess this was a safety feature that was introduced. It was also not [...]

Hello, how can I update Dana R firmware