nightscout/AndroidAPS #922: [3.0-beta5] DIA increase for FIASP

chrisbev, 2021-11-14:
Request please to increase DIA limit from 7 to say 8/9/10? I feel the need to try 7.5 or 8 for FIASP and there are various other Facebook posts with people making code changes/edits of 9.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

chrisbev, 2021-11-15:
Workaround:  Anyone searching for a workaround it has been brought to my attention that a pregnancy increases the max DIA to 10! Not suggesting getting pregnant or any advice but I am trying a Fake pregnancy in my preferences!

AdrianLxM, 2021-11-15:
I don't see any issues to set it to 10. The important part is that the peak is not off - and we restrict it in the settings to reasonable values.

MilosKozak, 2021-11-16:
https://github.com/nightscout/AndroidAPS/commit/f800b85f62fcfcca955a83d972f20d853fc8afe0

closed 2021-11-16 by MilosKozak