nightscout/AndroidAPS #923: [3.0-beta5] IOB prediction (Negative?) start value?

chrisbev, 2021-11-14:
Only seen with negative IOB but the IOB dark blue prediction line did not start at current IOB for me. Light blue appeared to. [3.0-beta5] Samsung A41 Android 11. See pic attached.
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2021-11-15:
basals are calculated up to next hour, prediction are always from "now"
what's wrong?

chrisbev, 2021-11-15:
Hi, on the picture the IOB is 0.68U and the light blue line starts there and decays to zero just off screen past 9:00hrs what I don't understand is why the dark blue line starts at about 0.4?  From the docs I get the impression the IOB predictions, light and dark blue, are for 0% basal and 100% basal?

MilosKozak, 2021-11-15:
one is calculated for current autosens result (light)
one for autosens 100% (dark)
but while checking this i decided to remove 100% prediction because it's more confusing than helping

closed 2021-11-15 by MilosKozak