nightscout/AndroidAPS #925: [3.0-Beta5] Eros beeps at unexpected moments without traceable alarm/alert

regiorunner, 2021-11-14:
Using Eros I heard an uexpected beep-beep (the same beepsound alert to warn when the pods expires).

The timing was strange:

Expiration reported by Pod screen: 13-11-2021 19:38

The beeps happened around (did not really log the precise timing, it can be +/- 10 minutes off):
12-11-2021 17:40
12-11-2021 18:40
12-11-2021 19:40
12-11-2021 20:40

When renewing the Pod screen after a beep there was immediate contact with the Eros, but no alarm or alert reported.
The beeping stopped as unexpected as it began, and the Eros ran 76 hrs without further issues.
I attached the log file of the relevant time period.

[AndroidAPS._2021-11-12_00-00-03_.1.zip](https://github.com/nightscout/AndroidAPS/files/7533890/AndroidAPS._2021-11-12_00-00-03_.1.zip)

samspycher, 2021-11-15:
@regiorunner I see a lot of "cancelling fake suspended TBR" messages. Could you perhaps send some log files from "normal" days?

regiorunner, 2021-11-16:
Thanks for the reply. The beep-beep problem only happened during the log file I send already. Please find attached another logfile. I checked already more logfiles, and the message  "cancelling fake suspended TBR" is frequent present in all logfiles.  But it could be only during that time there was a "overflow" causing the beep of the pod. Could this be related to the problem with the temporary basal, which always indicated "0" on the APS screen, also when the normal basal setting applies? I will check if I can find the number of that issue.

[AndroidAPS._2021-11-11_00-00-02_.0.zip](https://github.com/nightscout/AndroidAPS/files/7545351/AndroidAPS._2021-11-11_00-00-02_.0.zip)

regiorunner, 2021-11-16:
By the way, before the Eros I used a Dash. After the Eros I am using a Dash again, that seems to run without problems until now.

samspycher, 2022-05-21:
@regiorunner I was not able to discover the cause of the issue. If you are not re-experiencing it (if only because you switched to the Dash), I suggest we close this, because it does not seem to be an issue that occurs for other users.

closed 2026-10-15 by MilosKozak