nightscout/AndroidAPS #926: Security feature request : Stop insulin if near hypo

robertrub, 2021-11-14:
This subject has been often discussed in the Discord forums. Several users (me included) don't feel secure when we are near the hypo limit, a zero or small positif delta, and the algorithm decides to send insulin, justified for AAPS according to the calculations and predictions.

One part of the problem is _emotional stress_ : "I'm near hypo and I get injected insulin !" In a manual control system, none would inject insulin at those conditions.

The second part of the problem is _personal input errors_ or _misjudged food estimations_ leading to inexact predictions. AAPS predicts a BG rise but that rise may come later than predicted (or, may not come at all).

It would be very helpful to either have a security option or an automation with the logic of :

If _BG_ =< x AND _DELTA BG_ =< y THEN stop all insulin injections till next BG reading.

MilosKozak, 2021-11-15:
this is how the alg works
in default default installation there is a rule to set high TT if bg is below 4 mmol and SMB with high TT is disabled. This resolves issue you describe
If you change it it's up to you

closed 2021-11-15 by MilosKozak

Mendanbar, 2024-02-14:
I came across this whole discussion in [Discord](https://discord.com/channels/629952586895851530/629954570394533889/909449476249370705), and I'm also frustrated by the apparent lack of understanding.  Most of the answers to this problem (which seem to appear a lot) seem to be: "Your profile is not quite right" or "You need to use high temp targets" etc, but I think this is oversimplifying the situation.  We all know that the body and carbs together make for a complex situation.  We do our best to estimate, and we are all trying to dial in our profiles to be "correct".  But there are times where we need a safety net to prevent very real danger.  I fail to see how implementing an emergency insulin stop as a safety feature could be rejected.