nightscout/AndroidAPS #927: [3.0-beta5] Basal Prediction differentiator?

chrisbev, 2021-11-14:
I like the new predicted basal insulin predictor on the graph but could it be another colour to differentiate from past? Or say dotted outline like other predictors may be better please? [3.0-beta5] Samsung A41 Android 11.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2021-11-14:
why?

chrisbev, 2021-11-15:
To me initially when glancing at the graph I thought it was as previous i.e. the current state and not a prediction which is also not helped as the current time for me does not stand out very boldly.  Now after using for s few days I've learnt to look carefully!  Also thinking subsequently other predictions are dotted all-be-it they are curves?  A change would only help new users and I thought it may introduce the new feature to new users in a self explained way?

MilosKozak, 2021-11-15:
there is a "now" line to see where are we now
i don't plan to change this now

closed 2021-11-15 by MilosKozak