nightscout/AndroidAPS #930: AAPS restarts after attempting a profile change

ToreBj, 2021-11-14:
It took a while before I figured this one out. 
I had the following profiles:
[image]

I was editing profile "Normal week copy"  While changing the ISF value I changed my mind and wanted to go back to another profile. 
I pressed the profile pulldown meny and selected the first profile in the list. I then received a notification box asking if I wanted to discard the changes and change profile. I answered yes.

[image]

 After selecting the first profile "Normal week" I saw that the option for "Save" was still there when editing the first profile as if I had some unsaved cganges from editing the last profile.

[image]
.
 I made some changes to the "Normal week" profile and saved it. After that I selected the U200 profile. I made some edits and wanted to save and activate the U200 profile. AAPS restarts and U200 was not activated.

Summary:
After testing this several times I see that if you edit and discard the changes for a profile, that profile will be in illegal (Red) state.
Any profile before the illegal profile will be possible to activate, but the profile after the illegal profile can not be activated.

[image]

There's nothing in the log file, just a restart.

11:08:00.127 [main] D/CORE: [TimeListEdit.log():405]: 0: @00:00 10.0
11:08:01.570 [main] D/PROFILE: [LocalProfilePlugin.storeSettings():211]: Storing settings: {"defaultProfile":"U200","startDate":"2021-11-13T10:08:01.5680000Z","store":{"Normal week":{"dia":7,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":5}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":5}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0.9},{"time":"04:00","timeAsSeconds":14400,"value":0.9},{"time":"06:00","timeAsSeconds":21600,"value":1},{"time":"07:00","timeAsSeconds":25200,"value":1.1},{"time":"08:00","timeAsSeconds":28800,"value":1.1},{"time":"10:00","timeAsSeconds":36000,"value":1.05},{"time":"13:00","timeAsSeconds":46800,"value":1.05}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":5.5}],"target_high":[{"time":"00:00","timeAsSeconds":0,"value":5.5}],"units":"mmol","timezone":"Europe\/Oslo"},"Weekend":{"dia":7,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":5}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":2.5}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0.625},{"time":"04:00","timeAsSeconds":14400,"value":0.75},{"time":"06:00","timeAsSeconds":21600,"value":0.95},{"time":"07:00","timeAsSeconds":25200,"value":0.875},{"time":"08:00","timeAsSeconds":28800,"value":0.85},{"time":"10:00","timeAsSeconds":36000,"value":0.725},{"time":"13:00","timeAsSeconds":46800,"value":0.9},{"time":"22:00","timeAsSeconds":79200,"value":0.8},{"time":"23:00","timeAsSeconds":82800,"value":0.85}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":5.5}],"target_high":[{"timeAsSeconds":0,"value":5.5}],"units":"mmol","timezone":"Europe\/Oslo"},"Sensitive week":{"dia":7,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":4.7}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":3.8}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0.76},{"time":"04:00","timeAsSeconds":14400,"value":0.81},{"time":"06:00","timeAsSeconds":21600,"value":0.9},{"time":"07:00","timeAsSeconds":25200,"value":0.97},{"time":"08:00","timeAsSeconds":28800,"value":0.92},{"time":"10:00","timeAsSeconds":36000,"value":0.85},{"time":"13:00","timeAsSeconds":46800,"value":1},{"time":"16:00","timeAsSeconds":57600,"value":1.06},{"time":"17:00","timeAsSeconds":61200,"value":1.06},{"time":"20:00","timeAsSeconds":72000,"value":1.06},{"time":"22:00","timeAsSeconds":79200,"value":1.06}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":5.5}],"target_high":[{"timeAsSeconds":0,"value":5.5}],"units":"mmol","timezone":"Europe\/Oslo"},"LocalProfile1":{"dia":5,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":0}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":0}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":0}],"target_high":[{"time":"00:00","timeAsSeconds":0,"value":0}],"units":"mmol","timezone":"Europe\/Oslo"},"Normal week copy":{"dia":7,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":4.7}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":3.8}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0.85},{"time":"04:00","timeAsSeconds":14400,"value":0.9},{"time":"06:00","timeAsSeconds":21600,"value":1},{"time":"07:00","timeAsSeconds":25200,"value":1.08},{"time":"08:00","timeAsSeconds":28800,"value":1.02},{"time":"10:00","timeAsSeconds":36000,"value":0.88},{"time":"13:00","timeAsSeconds":46800,"value":1.06},{"time":"16:00","timeAsSeconds":57600,"value":1.06},{"time":"17:00","timeAsSeconds":61200,"value":1.06},{"time":"20:00","timeAsSeconds":72000,"value":1.06},{"time":"22:00","timeAsSeconds":79200,"value":1.06}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":5.5}],"target_high":[{"timeAsSeconds":0,"value":5.5}],"units":"mmol","timezone":"Europe\/Oslo"},"LocalProfile2":{"dia":5,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":0}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":0}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":0}],"target_high":[{"time":"00:00","timeAsSeconds":0,"value":0}],"units":"mmol","timezone":"Europe\/Oslo"},"LocalProfile3":{"dia":5,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":0}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":0}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":0}],"target_high":[{"time":"00:00","timeAsSeconds":0,"value":0}],"units":"mmol","timezone":"Europe\/Oslo"},"U200":{"dia":7,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":9}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":10}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0.55},{"time":"06:00","timeAsSeconds":21600,"value":0.6},{"time":"08:00","timeAsSeconds":28800,"value":0.55}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":5.5}],"target_high":[{"time":"00:00","timeAsSeconds":0,"value":5.5}],"units":"mmol","timezone":"Europe\/Oslo"},"LocalProfile5":{"dia":5,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":0}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":0}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":0}],"target_high":[{"time":"00:00","timeAsSeconds":0,"value":0}],"units":"mmol","timezone":"Europe\/Oslo"}}}
11:08:01.601 [RxCachedThreadScheduler-54] D/CORE: [UserEntryLogger$log$2.invoke():44]: USER ENTRY: STORE_PROFILE LocalProfile  [SimpleString(value=U200)]
11:08:03.498 [main] D/CORE: [MainApp.onCreate():71]: onCreate
11:08:03.504 [main] D/CORE: [MainApp.onCreate():97]: Version: 2.8.2.10-dev
11:08:03.504 [main] D/CORE: [MainApp.onCreate():98]: BuildVersion: NoGitSystemAvailable-2021.11.12-12:57
11:08:03.504 [main] D/CORE: [MainApp.onCreate():99]: Remote: NoGitSystemAvailable

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2021-11-15:
fixed save button behavior https://github.com/nightscout/AndroidAPS/commit/6aaedefbed14289f8965d402e0563400fdbc6d2d
but unable to reproduce crash or red state like you desrcibe

ToreBj, 2021-11-16:
Ok, set ISF to 0.1 or zero, i.e. illegal and you have a red screen. Then switch to another profile.

MilosKozak, 2021-11-18:
i disabled switching if profile is invalid. i still cannot simulate crash https://github.com/nightscout/AndroidAPS/commit/284dc19bd16741ddb013c43b819e8ea2ff6a0240
validity of profile is checked when you try to switch to profile. this is still in place

closed 2021-11-21 by MilosKozak