nightscout/AndroidAPS #932: [2.8.2] Failed meal bolus queued and administered along with the reentered one

ireneusz-ptak, 2021-11-15:
AAPS 2.8.2 + Accu-Chek Combo

While using the Bolus Wizard the SMB was administered and it caused the meal bolus to fail. The caretaker checked the pump for bolus history and the last one was 0,1u, so she reentered the meal and AAPS did send both meal boluses to the pump.

Queuing of the failing meal bolus (instead of leaving the progress bar in focus or actually removing the bolus from the queue) looks like a terrible design decision.

SMB - 0,1u
1st meal bolus - 1,4u
2nd meal bolus - 1,3u

Started around 13:43:21:
[AndroidAPS._2021-11-08_00-00-00_.14.zip](https://github.com/nightscout/AndroidAPS/files/7537394/AndroidAPS._2021-11-08_00-00-00_.14.zip)

andyrozman, 2021-11-15:
This sometimes happens... SMB bolus "hijacks" progress dialog sometimes, but your placed bolus still happens... You just have to wait and it will appear in Combo tab.

MilosKozak, 2021-11-15:
* executing of SMB started
13:43:37.616 [Thread-28802] D/PUMPQUEUE: [QueueThread.run():137]: performing SMB BOLUS 0,10 U
* bolus wizard confirmed
13:43:44.689 [main] D/CORE: [BolusWizard$commonProcessing$1.run():431]: USER ENTRY: BOLUS WIZARD insulin 1.4 carbs: 40.0
* SMB finished
13:43:48.710 [Thread-28802] D/DATATREATMENTS: [TreatmentService.createOrUpdate():406]: New record from: PUMP Treatment{date= 1636375320000, date= 08.11.2021 13:42, isValid= true, isSMB= true, _id= null, pumpId= 1636375320000, insulin= 0.1, carbs= 0.0, mealBolus= false, source= 1}
* bolus delivering started
13:43:48.829 [Thread-28802] D/PUMPQUEUE: [QueueThread.run():137]: performing BOLUS 1,40 UCARBS 40 g
* bolus wizard confirmed
13:44:40.881 [main] D/CORE: [BolusWizard$commonProcessing$1.run():431]: USER ENTRY: BOLUS WIZARD insulin 1.3 carbs: 40.0
* error dialog displayed "Polecenie jest teraz wykonywane"
13:44:40.990 [main] D/CORE: [ErrorDialog.onCreateView():49]: Error dialog displayed
* delivering of bolus finished
13:44:54.297 [Thread-28802] D/DATATREATMENTS: [TreatmentService.createOrUpdate():406]: New record from: PUMP Treatment{date= 1636375392000, date= 08.11.2021 13:43, isValid= true, isSMB= false, _id= null, pumpId= 1636375392999, insulin= 1.4, carbs= 40.0, mealBolus= true, source= 1}
* bolus wizard confirmed
13:45:03.505 [main] D/CORE: [BolusWizard$commonProcessing$1.run():431]: USER ENTRY: BOLUS WIZARD insulin 1.3 carbs: 40.0
* started delivering
13:45:03.532 [Thread-28841] D/PUMPQUEUE: [QueueThread.run():137]: performing BOLUS 1,30 UCARBS 40 g
* bolus finished
13:45:52.011 [Thread-28841] D/DATATREATMENTS: [TreatmentService.createOrUpdate():406]: New record from: PUMP Treatment{date= 1636375452000, date= 08.11.2021 13:44, isValid= true, isSMB= false, _id= null, pumpId= 1636375452000, insulin= 1.3, carbs= 40.0, mealBolus= true, source= 1}

MilosKozak, 2021-11-15:
from what i see AAPS behavior was correct
first attempt of 2nd bolus was refused because previous bolus was still running
second attempt of 2nd bolus was executed because if was after finish of 1st bolus

closed 2021-11-15 by MilosKozak

ireneusz-ptak, 2021-11-16:
@MilosKozak Thanks for checking - it seems the situation was described to us differently.

ireneusz-ptak, 2021-11-16:
On the other hand why the 1.4 bolus dialog disappeared from the screen allowing to enter the meal 2nd time?

MilosKozak, 2021-11-16:
it can be problem described from andy. it will be improved in 3.0