nightscout/AndroidAPS #934: 2.8.2.10_dev / BYODA  calibration loops

robertrub, 2021-11-15:
AAPS, last dev. Had the same problem with the prior dev too. BYODA is the BG collector from G6 transmitter.

If I select _calibration_ in BYODA menu, it works ok.

In AAPS, if I hit the _calibrate_ button on the main screen, BYODA opens on its calibration screen. That's when things go sour. BYODA enters an infinite loop and asks the calibration number over and over. 

Each time I enter the blood BG in BYODA, it creates an entry in AAPS _treatments_ too.

Looks like that BYODA is doing its job but AAPS keeps sending the user back to the calibration screen of BYODA.

MilosKozak, 2021-11-15:
i confirm the issue
you have press twice back to exit the calibration dialog
but i don't know how to fix it. It's cheating with dexcom app anyway. it depends on in internal app processing
you may ask BOYDA devs

closed 2021-11-15 by MilosKozak

robertrub, 2021-11-16:
@MilosKozak OK, I understand the origin of the problem is not ours but something is doable ;) 

Either you can make this button unavailable for Dexcom patched (if the problem is for both BYODA and patched Dexcom app) or show a pop-up message when the button is tapped "Once you have entered your BG in Dexcom, tap twice the "back" button of Android" so that users will know what to do.