nightscout/AndroidAPS #939: [dev] Manual profile switches sent twice to pump

Philoul, 2021-11-15:
It seems to be a recent bug...

At 3:00:48 I installed a previous built done after https://github.com/nightscout/AndroidAPS/commit/081e8bd684632cfa795b68dd623f9663fe962d2d (10/11 9h24) and PS is sent twice to pump

Then I installed at 3:05:42 an older built done after https://github.com/nightscout/AndroidAPS/commit/22fced61fe72870f9891d2a9cada081a8d5244eb (08/11 13:37) and PS was sent only once to pump

Just one trial it can be luck...

[image]

MilosKozak, 2021-11-15:
this is the reason. profile name checking 
 https://github.com/nightscout/AndroidAPS/commit/b4dec7195c028a987974eaf3a8e3ca0acc37929b#diff-7bb20fd96aa8b14c4f8761da416e727b07c4ab8cad22e8d4b9bea6aaf398b0cc

robertrub, 2021-11-16:
Isn't this the same thing as https://github.com/nightscout/AndroidAPS/issues/827#issue-1040934646  ?

MilosKozak, 2021-11-16:
i prepared a solution but i have to test it first

MilosKozak, 2021-11-16:
it's a duplicate. this seems to independed to pump driver

MilosKozak, 2021-11-16:
fixed by https://github.com/nightscout/AndroidAPS/commit/f48dcc77e23d499ab0c1af8499ce9ce8ed574ba1

closed 2021-11-16 by MilosKozak