nightscout/AndroidAPS #943: [3.0-beta6] NSClient Crash on startup

shafiro, 2021-11-16:
NSClient 3.0-beta6
BuildVersion: ae52998d2-2021.11.16-21:20
Flavor: nsclient-debug/release
Nightscout version: 14.2.4
Engineering mode enabled
Device: Xiaomi MI 9, Android 11, MIUI 12.5.1

NSClient crashes on startup ,a few seconds after logo shows up.

PlayProtect is Off.
Tried Uninstall, reinstall, force stop, clear data (android settings), delete AAPS folder, delete project folder, build from different pc.

NSClient 3.0-beta5 works.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7549784/AndroidAPS.log)

streethawk22, 2021-11-16:
I have exactly the same experience...NSClient crash on startup...I have Galaxy Note10+. All others 3.0 Beta versions without Problems.

shafiro, 2021-11-17:
@streethawk22
Please upload log

streethawk22, 2021-11-17:
I can't now upload log, then when the App not starts,  she have's not the logs directory and I can't go to the App and Send Logs per Email.

shafiro, 2021-11-17:
If you give storage permission via android settings, you will have the log.

streethawk22, 2021-11-17:
Yes, this have I not tried...THX

Here is the LOG file.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7554068/AndroidAPS.log)

MilosKozak, 2021-11-19:
could you try dev?

shafiro, 2021-11-19:
No more crash with current dev. thanks a ton!

closed 2021-11-20 by shafiro