nightscout/AndroidAPS #944: Duplicate Glucose

pbloom209, 2021-11-17:
Dana I
Using Beta3 ver 6
Xdrip. but prefer Dexcom.
Everything was working properly.
Sensor fellout so I needed to insert a new one.
Immediately using Xdrip got duplicate glucose symbol.
Uninstalled Xdrip and installed Dexcom.
Still have duplicate readings.
Previously I fixed by resetting database.
Starting fresh without pulling in data from Nightscout.

[image]
[image]
[image]

streethawk22, 2021-11-17:
Try to manual delete the duplicate BGs on AAPS, It can Help.

swissalpine, 2021-11-17:
I've seen the same behaviour some days agos: After a Sensor Stop (Dexcom G6, xDrip+) all BG readings are duplicated. This workaround is a solution for the moment - but something strange happens ...

MilosKozak, 2021-11-17:
it's a bug on xDrip side we cannot resolve. xDrip sends data from previous session with different timestamps
in release notes is already mentioned you have to switch to dexcom app and do not use xDrip+ as a dexcom collector for AAPS

closed 2021-11-17 by MilosKozak

pbloom209, 2021-11-17:
I did also observe using Dexcom, the calibration entry flowed into AAPS and casued another duplicate BG, it was easy to fix as it was just one entry,

MilosKozak, 2021-11-17:
@pbloom209 if this happen again please provide logs and timestamp of removed entry