nightscout/AndroidAPS #946: [3.0-beta6] Bolus incerements discrepancy

DM-LTR, 2021-11-18:
Regardless of the value of bolus step set in pump / Dana tab, there is fixed value of 0,05 U in Bolus Calculator / Insulin button "+ sign" / bolus_increment (in SMB tab)

[image]
[image]
[image]
[image]
[AndroidAPS._2021-11-17_15-19-17_.0.zip](https://github.com/nightscout/AndroidAPS/files/7561138/AndroidAPS._2021-11-17_15-19-17_.0.zip)

martinlcz, 2021-11-18:
I have same problem. Dana RS v3

DM-LTR, 2021-12-16:
Issue still present even in beta 11, last working version was beta 5.
Regardless of bolus increment set in pump (0,1 U) bolus increment in AAPS is fixed at 0,05 U

pmeder, 2022-02-08:
I have updatet to AAPS 3.0.0.1 since that update there is a problem with bolus delivery. The bolus calculator uses numbers with 2 digits after the comma value. That leads to an failure when trying to give a bolus with lets say 1,55 if the bolus is 1,50 no failure appears. I think the failure is the same as discribed above, or should I open a new issue for that?

martinlcz, 2022-02-08:
You must set bolus step 0.05  (no 0.1) (in doctor menu)

MilosKozak, 2026-10-13:
Fixed in `dev` with 951a01d81e.

**What was wrong:** the Dana driver reads the bolus step and the basal step from the pump, but it wrote them into a pump description that is built again every time it is read. So the steps were lost, and AAPS always used the step of the pump type (0.05 U for Dana RS). With the pump set to 0.1 U, AAPS offered amounts like 1.55 U, and the pump refused them.

**Now:**
- Once the steps were read from the pump, they are in the pump description. The bolus wizard, the insulin dialog "+" buttons, QuickWizard, the safety checks and `bolus_increment` for SMB all use them.
- Before the steps are read (or after a refresh), the steps of the pump type are used, as before.
- The workaround of setting the pump to 0.05 U is no longer needed.

Tested with unit tests, not yet with a real pump. If you still see 0.05 U steps on a pump set to 0.1 U with the current `dev`, please reopen and attach a log.

closed 2026-10-13 by MilosKozak