nightscout/AndroidAPS #947: [enhancement] Allow us to perform arithmetics in certain numeric fields

m-scha1337, 2021-11-18:
A pretty minor feature request i believe. 

I particularly think of this improvement almost every time when i need to set a "Duration" (in minutes) in the "Temporary Target" submenu. Currently, when you want to set a Temporary Target for, say, 6 hours (something i might want to do when i go to bed in order to keep me extra clear of hypos), you would either have to switch to a calculator app, enter 60\*6 and carry over the result back to AAPS, or do 60\*6 in your head. Don't get me wrong, that's absolutely doable and not a big challenge to ask for.
But it would still be a minor improvement to the entire AAPS workflow if we were able to just put "60\*6" into the "Duration" field in the "Temporary Target" submenu and it would just calculate the result then and there. 
Aside, i believe this would be quite trivial to implement.
In addition, this would improve the AAPS workflow for children, disabled individuals or just people who aren't good at mental maths immensely.

I also believe this would be beneficial for the "carbs" field in the bolus wizard, since oftentimes you may eat a meal of which you only know the individual carb portions of, not however the total amount. Meaning you had to add up however many gram carbohydrates you're eating in either your head, or again, use an external calculator to add all individually known gram portions together into a total result.
For example: Being able to put in "(30+29)\*2" because you are going to eat a McDonalds Hamburger (30g) and small fries (29g) twice is just far less cumbersome in my opinion than doing it in your head, especially if you're about to eat and perhaps are in a time rush.. 

There may also be more fields where this might come in useful..

---

A good example of how this is already implemented in software is "adobe illustrator" (or similar), where you can do "Y: 50px+100px" to shift an object 100px in the Y dimension.

CaptainBalou, 2021-11-18:
I would love this too.

StephenBrown2, 2021-11-18:
The hours vs minutes fields for time shift vs duration of profile switches have also messed me up so many times that I've modified my app to simply never display the "time shift" field. Has cleared things up well for us.

savek-cc, 2021-11-18:
https://medium.com/coding-blocks/making-arithmetic-parser-with-kotlin-4097115f5af

szantos, 2021-12-21:
> I particularly think of this improvement almost every time when i need to set a "Duration" (in minutes) in the "Temporary Target" submenu. Currently, when you want to set a Temporary Target for, [...]

For the time part of your suggestion:
IMHO in many cases you want to be able to set the end_time as for example 6:00am

MilosKozak, 2026-10-15:
recalculated in v4

closed 2026-10-15 by MilosKozak