nightscout/AndroidAPS #948: [3.0-beta6] No Messages but Alarm by empty Battery and empty Reservoir

69nobody, 2021-11-18:
Please excuse me for my bad english.
Version Beta6
the reservoir is empty - but there are no messages only i hear a signal ton
the battery is empty - but there are no messages only i hear a signal ton
aaps show a red triangle on the left side on the top. when i push the triangle the following information was displayed.
Blood values are to near?!

69nobody, 2021-11-18:
addional the signal ton can only turn off when you restart your smartphone.

MilosKozak, 2021-11-18:
what pump?

69nobody, 2021-11-18:
Dana i
should i send teh log files?

MilosKozak, 2021-11-18:
when low battery or low reservoir accures aaps displays notification
pump will beep

69nobody, 2021-11-18:
I have seen no notification

MilosKozak, 2021-11-18:
i tested dana-i and can confirm that pump is not sending notification to phone
but it beeps regulary
on AAPS side you'll see red status light for reservoir on overview

red triangel is related to bad CGM data

closed 2021-11-18 by MilosKozak