nightscout/AndroidAPS #950: Lost carb entry by 2.8.2 after using calculator

mk886714, 2021-11-18:
I entered a meal of 24g carbohydrates using the bolus calculator.  As my blood sugar continued to rise over the next few hours, I noticed that the insulin release had been entered, but an entry for the carbohydrates was missing.  this set TBR0 and therefore the BZ rise.  maybe this should be tested somehow
[image]
[image]

--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

mk886714, 2021-11-18:
[image]

MilosKozak, 2021-11-18:
2.8.2 is at the end of it's life. there will no fixes added because code of 3.0 is by 50% new
test with 3.0 and open issue if it appears again

MilosKozak, 2021-11-18:
thanks for reporting anyway

closed 2021-11-18 by MilosKozak