nightscout/AndroidAPS #951: SMB are converted to meal bolus on app restart

savek-cc, 2021-11-18:
When restarting AAPS (it's enough to end the app and start it again), all past SMB are converted to meal bolus in the graph.
AAPS 2.8.2.11-dev 31d6ca2f9-2021.11.17 etc.

Before app restart:
[image]
After restart while still reading pump history
[image]
After completing initialization
[image]

MilosKozak, 2021-11-18:
guessing you have xDrip synchronized with NS

savek-cc, 2021-11-18:
REST-API enabled, Send Display Glucose enabled, Download treatments enabled.
Wrong/incompatible settings?

savek-cc, 2021-11-18:
AAPS NS client: upload enabled, download profiles enabled; all other options are disabled.

MilosKozak, 2021-11-19:
my experinece is that xDrip modify data and send back to NS.
or better accepts data from NSClient (which is NS data aready) modify and upload to NS
this data is synced back to AAPS from NS and this leads to behavoir you observe now

savek-cc, 2021-11-19:
But why does AAPS take anything other than the configured items from NS? I only enabled:
- upload
- download profiles
Why does that affect the type of bolus? Apart from profiles (which I could disable now as it's converted to a local one) AAPS should not be downloading/syncing anything from NS?

MilosKozak, 2021-11-19:
hard to say and probably not a worth to spend time with digging in logs because in 3.0 this parts of code are completely new

savek-cc, 2021-11-19:
? This is current dev - i.e. 3.0.

MilosKozak, 2021-11-19:
ahhhh ok. then when this happen again upload logs pls

swissalpine, 2021-11-19:
I see this too and don't upload treatments to xDrip+, only from AAPS to Nightscout and vice versa. Next time I restart the app will share a log.

swissalpine, 2021-11-19:
Perhaps these are reimported from the pump. Because at first the triangles are visible, then, during the pump initialization, they disappear and numbers appear:
[image]
[image]

Log of the restart
[Log.txt](https://github.com/nightscout/AndroidAPS/files/7571214/Log.txt)

swissalpine, 2021-11-19:
By the way: The message "Pump driver changed" isn't correct. Problem seems for me that the Omnipod driver is initialized (?) befor my real pump Spirit Combo.

savek-cc, 2021-11-19:
It really happens on reading the pump history on app start. See two screenshots. First is before stopping AAPS - SMB all the way back to last night. Pump history (checked on pump) has entry 30/30 (oldest one) at 6:02.
After AAPS start, the SMB are first still visible as SMB - until the history is read from the Combo - at which point they are converted to regular boli. **But only the ones that are present in the combo's internal history**.
Screenshots and log attached.

[image]
[image]
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7571290/AndroidAPS.log)

MilosKozak, 2021-11-19:
it was combo specific issue. fixed by https://github.com/nightscout/AndroidAPS/commit/f5faa3492288b97b49284e116c29720f50194560
thanks!

closed 2021-11-19 by MilosKozak