nightscout/AndroidAPS #952: [3.0 beta6] combo suddenly warns about cancelled tbr

RalfCGM, 2021-11-19:
Dont understand exactly whats happening. But in 8 hours the combo gave 4 times a message about TBR cancelled. Message i didnt have with beta5. 

Will provide Logs and time if it happens again.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

closed 2021-11-19 by MilosKozak

RalfCGM, 2021-11-19:
Is this issue already resolved? I would love to provide additional information if nescessary.

MilosKozak, 2021-11-19:
create a new issue with logs
keeping too many open issues in state where we cannot help is hard to manage