nightscout/AndroidAPS #953: Dev / BYODA v1.9 : Calibration in BYODA give double readings in AAPS

robertrub, 2021-11-19:
[image]

I have installed the new version 1.9 of BYODA. All was working OK till I got the red triangle when I calibrated this morning. I saw that in the Dexcom tab, there were 2 same readings at the time of the calibration

[image]

[image]
 
Deleting of the readings cancelled the red triangle

[AndroidAPS_LOG_1637304353677.log.zip](https://github.com/nightscout/AndroidAPS/files/7568517/AndroidAPS_LOG_1637304353677.log.zip)
.

MilosKozak, 2021-11-19:
2nd timestamp is not visible . do you still have it?

swissalpine, 2021-11-19:
Could AAPS delete double entries when they come in (condition for example: (same bg, same delta, timestamp < 60 sec)?

MilosKozak, 2021-11-19:
there is a wide range of CGM producing different data. so far i'm trying to find setup which doesn't produce duplicates instead of cheating/filtering on AAPS side (which may work for one CGM but not for others)

robertrub, 2021-11-20:
> 2nd timestamp is not visible . do you still have it?

Sorry, nope. I hadn't realized that the text was longer than the windows. 

It could be better f the text was shorter and more explicit.... ;)

vanelsberg, 2021-11-20:
I can see why upgrading BYODA could result in duplicates:
The new version does not have any history of it predecessor: it just read the memory of the transmitter containing several hours of BG data and dump all to APPS?

robertrub, 2021-11-21:
@vanelsberg The duplicate issue I opened was for _calibration_, not for general use.

To answer your question: i changed from 1.7 to 1.9 towards the end of the 2h wait period (no BGs for 2h) of a sensor restart and I didn't have any duplicates.

closed 2021-11-21 by MilosKozak

robertrub, 2021-12-04:
@MilosKozak  More testing on BYODA calibration (v1.9) : I did the same things at +3 min after last BG reading at at +4 min after last reading and the first did NOT give a double BG value and the 2nd did give a double value. Several messages on Discord gives the same impression of this bug being time related to the last BG reading.

Open BYODA by its own icon, calibrate in BYODA and see what happens in AAPS. The attached log with double BGs in AAPS. Look around 8:48

[image]

[AndroidAPS_LOG_1638604271854.log.zip](https://github.com/nightscout/AndroidAPS/files/7653603/AndroidAPS_LOG_1638604271854.log.zip)

vanelsberg, 2021-12-04:
> ... trying to find setup which doesn't produce duplicates instead of cheating/filtering on AAPS side ...
How about implementing some threshold allowing "N" duplicates within a set time window (e.g 1h)?

This way it would allow casual duplicates caused by calibration or manually entering, but alert on multiple duplicates generated consistently (indicating some sort of automation or alternate BG source)