nightscout/AndroidAPS #954: DEV/ BYODA v1.7 : Calibration date error in _treatments_ tabs

robertrub, 2021-11-19:
I have restarted a sensor in BYODA v1.7. All went ok but I see here is an date error in one of the _Treatments_ tab (date 2106...) while on the other, the date is correct.

[image]

[image]
 
[image]

MilosKozak, 2021-11-19:
session was started in BOYDA 1.9 and the downgraded to 1.7?

robertrub, 2021-11-19:
No, sensor endod with v1.7, waited 15+ minutes then restarted with v1.7. Uninstalled v1.7 15-20min before the end of the 2 hour waiting time and installed v1.9 (didn't wan to clean too many duplicates if I was bitten with the problem).  First BGs arrived via v1.9 after the wait time finished.

MilosKozak, 2021-11-19:
hmmm nobody knows what dexcom app does. aaps only accepts sent data. if they come with wrong timestamps ....
open new issue when this happen after you do whole procedure with 1.9

closed 2021-11-19 by MilosKozak