nightscout/AndroidAPS #958: Dash: remove the 'suspend delivery' button

avereha, 2021-11-20:
I'm opening this issue so we could discuss the idea of removing the 'suspend delivery' button for Dash.
'Disconnect pump' does set a zero TBR and has the advantage that delivery will be resumed automatically.

Is there are any good use-case or reason to keep the 'suspend delivery' button?

robertrub, 2021-11-20:
_Disconnect pump_ is safer as it turns basals back on after x minutes. Can't see where more than 3h of disconnect would be utilized.

_Suspend button_ can be deleted from the DASH tab.

vanelsberg, 2021-11-20:
Was using the suspend button in favor of "disconnecting the pump" from AAPS.
I do feel the suspend is not identical with "disconnect": warning every 15 minutes, letting Dash fall  back to default basal rate.

However, after some discussion, I do think removing the suspend/resume buttons from the DASH driver is a good idea:
* The driver suspend is conflicting with Profile switch as you pointed out.
* Standard AAPS functionality is on the main overview. Having an additional ways to "disconnect" adds confusion.
* AAPS's "disconnect" automatically "reconnects" the pump after 3 hours (@robertrub)

So I would say you have a sound reason on removing suspend/resume.

robertrub, 2021-11-21:
@avereha  Be sure to keep the "Resume" delivery function/button though. If a profile change doesn't get trough and the pod stays "stuck" in suspended" mode, the user should be able to "Resume" insulin delivery.

avereha, 2021-11-25:
This can be closed

closed 2021-12-18 by avereha