nightscout/AndroidAPS #961: [3.0-beta6] Loop unresponsive/disabled

CaptainBalou, 2021-11-20:
I did my first interaction with AndroidAPS that day and realized a few things.

1. It was 09:31 am but the diagram showed something around 00:50 am
2. I had no BG value since 523 minutes
3. A minute after maximizing the app an alert of 11:21 pm past night came up
4. I had no BG values in BYODA

I use a Huawei P20 Pro which sometimes has an issue with BT driver. Just a reboot of the device can fix this. So I restarted the cell phone. After that I realized this:

1. I got BG values again
2. Loop was currently disabled
3. Loop was deactivated in configuration section

After enabling loop in config section all was fine again and is running with no errors since then. But having a disabled/deactivated loop was the first time for me and doesn't look right to me.

(Sorry for the German translation in the screenshots but I'm not able to keep AndroidAPS in English permanently)

[AndroidAPS._2021-11-20_09-35-22_.0.txt](https://github.com/nightscout/AndroidAPS/files/7575473/AndroidAPS._2021-11-20_09-35-22_.0.txt)

[image]

[image]

[image]

[image]

MilosKozak, 2021-11-21:
already fixed in dev https://github.com/nightscout/AndroidAPS/commit/d9f68bca3fb89b1b8ffe89d9ade5e544e0e84e5d

closed 2021-11-21 by MilosKozak

CaptainBalou, 2021-11-21:
Hello @MilosKozak thanks for replying. I probably haven't got the order of beta and dev builds right. I thought during beta test phase all is tested in the beta and fixed in the next beta. Haven't thought it could be fixed in dev already. So the correct order would be to always see if the error I've seen is already fixed in dev. But reproducing is sometimes difficult. 😀 I try to pay more attention on the dev branch during tests of betas. 

Sorry for generating extra load with this ticket.