nightscout/AndroidAPS #969: Failed basal notifications don't adhere to Local Alert Settings

rustymonkey, 2021-11-21:
In order to limit the intrusiveness of failed basal alerts, I tried adjusting alert settings but experienced unexpected behavior re. two settings:

1) I lowered the media volume to 0 with vibration enabled, and set Preferences > Local Alerts > Advanced > 'Use system notifications for alerts and notifications':ENABLED. With this, I would have expected that failed basal updates would generate a system notification/vibration, however, the resulting alerts don't trigger notifications / vibrations as would be expected. Instead they trigger an audio alarm that is controlled by Media volume rather than Notification volume (i.e. the alert isn't generated as an Android system notification as per the configuration).

2) In a similar fashion, setting Preferences > Local Alerts > Advanced > 'Alert if pump is unreachable' to DISABLED, to suppress these alerts. However, disabling this has no effect--failed basal adjustment alerts continue to appear even if this is disabled. Is this setting only meant to generate an alert when there's no associated failed bolus / basal adjustment?

Note: 
- Tested with 3.0 Beta 6, with a Medtronic 554 connected to AAPS running on a Galaxy S7/Android 11 (Android Experience) via an Orangelink (2.5 firmware)
- Related to: #889 , #884

MilosKozak, 2021-11-24:
there are different things
* Notifications for events where resolution can be postponed, messages etc
* Errors what can lead to serious problems ie what if cancel of TBR fail and your BG is heading double arrow down and running TBR is 500% ?? You sent bolus, locked phone and bolus was not successful ??
* Pump unreachable when it's not pump is not available for longer time (btw should not be disabled because of the same reason)

Your system must be reliable enough to not produce errors

closed 2021-11-24 by MilosKozak

rustymonkey, 2021-11-25:
Perhaps following local alert settings may not be the ideal approach in all cases (though the UI could use some wordsmithing to make it more understandable which Alerts follow which settings). However, if the objective is for the alerts to trigger users to address urgent issues immediately, the current design doesn't accomplish this as effectively as it could:
1) It doesn't generate an associated vibration. So if the alarm sound is low or the user is in a noisy environment, the user isn't notified.
2) The current design encourages users to turn the volume down to near 0 when they're in situations in which they cannot disturb others (e.g. presenting to a group / sitting in a lecture / etc.). Considering the absence of an associated vibration, this is the opposite outcome of what is desired.
3) The current design encourages users to turn the volume down because it can be triggered in numerous scenarios that are not urgent and do not require immediate action. For example:
- transient SMB failure
- transient failure to raise basal
- transient failure to lower basal when BG is very high