nightscout/AndroidAPS #972: [3.0beta6] AAPS stopped

chrisbev, 2021-11-22:
While waking at about 4:00hrs on looking at phone AAPS Started (may have been stopped) popped up and no BG were in the graph for about the last 40mins.
Previous interaction with AAPS: At about 2:30hrs I woke and noticed my BGs were running flat but above target with positive IOB.  Earlier the previous evening I was recovering from a period of high BG, faulty canular and FIASP water!  I did a temporary reduced target and a correction (using prime as IOB and running flat but high BG).
At and since 4:00hrs AAPS seems 'normal.
First 24hrs of beta6 so too early to see if a pattern but this has not previously happened with beta5 or 2.8.2.
Logs sent by email.
FSL2 patched Librelink xDrip Samsung A41 Android 11.
Just noticed xDrip did not have any BG for the 40mins prior to 4:00hrs which to me leads me to think it could be a phone problem or did say AAPS cause a 'phone problem?
[image]
[image]
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2021-11-22:
03:23:12.942 AAPS stopped
04:02:58.644 woke up

but no crash or reason visible in logs, sorry

MilosKozak, 2021-11-22:
but seems to phone issue rather than AAPS because new BG would wake up AAPS

closed 2021-11-22 by MilosKozak