nightscout/AndroidAPS #975: [3.0-beta6] Add Basal Daily Total to Profile

chrisbev, 2021-11-22:
Request that the total of the daily basal is added to the profile summary found by pressing the profile in the top bar of the home screen.
Reason: Other profile screens show the daily basal total and I find when checking that any basal changes I have made are correct the basal total is a good confidence, the other parameters are reasonably easy to remember but each of the 24 hourly values I find difficult.  When making changes I like to check back at that screen that the changes were implemented as I intended.
[3.0-beta6] Samsung A41 Android 11.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Philoul, 2021-11-23:
See proposal in PR #978 (I added a screenshot)

chrisbev, 2021-11-23:
Thanks @Philoul the screenshot looks great.  I expect it is very low risk but will try to build the PR soon...

closed 2021-11-23 by MilosKozak

Philoul, 2021-11-23:
Oups, crash detected in CommandTempBasalPercent (strange because it's only when I disconnect pump)

PR #978 (2021-11-23): Add basal daily total to profile viewer
close #975

[image]