nightscout/AndroidAPS #976: Bolus registred under treatments but not delivered

amdaini, 2021-11-22:
We had a serious issue this morning; my daughter bolused 2.80u for breakfast, one hour later I got a call from the school nurse as his bg was unexpectedly raising. I checked on the  nightscout and there was the 2.8u registered, also checked on the treatment tab of his AAPS and the 2.8u was also there but on the pod history strangely I was unable to find the breakfast bolus. 
AAPS version: 2.8.2.10-dev
Pump: Omnipod Dash 
CGM: Dexcom G6/xdrip+
PFA to this issue the AAPS logs and some screenshots of the treatments tab and pod history:

[AndroidAPS._2021-11-22_00-00-02_.2.zip](https://github.com/nightscout/AndroidAPS/files/7584613/AndroidAPS._2021-11-22_00-00-02_.2.zip)

[image]
[image]
[image]

avereha, 2021-11-22:
This is **not** dash specific.
It is also not a bug, but maybe we can improve the 'Treatments' button.

After reading the logs, it looks like the entry was added via the 'Treatments' button and the 'Do not bolus, record only' option was checked:
```
08:07:44.204 [RxCachedThreadScheduler-68] D/CORE: [UserEntryLogger$log$2.invoke():44]: USER ENTRY: TREATMENT TreatmentDialog Enregistrement [SimpleString(value=Record), Insu
lin(value=2.8), Gram(value=28)]
08:07:44.238 [RxCachedThreadScheduler-60] D/DATABASE: [TreatmentDialog.submit$lambda-13$lambda-12$lambda-5():160]: Inserted bolus Bolus(id=557, version=0, dateCreated=1637564864207, isValid=true, referenceId=null, interfaceIDs_backing=null, timestamp=1637564864192, utcOffset=3600000, amount=2.8, type=NORMAL, isBasalInsulin=false, insulinConfiguration=null)
08:07:44.241 [RxCachedThreadScheduler-35] D/DATABASE: [TreatmentDialog.submit$lambda-13$lambda-12$lambda-8():166]: Inserted carbs Carbs(id=260, version=0, dateCreated=1637564864232, isValid=true, referenceId=null, interfaceIDs_backing=null, timestamp=1637564864192, utcOffset=3600000, duration=0, amount=28.0)

```

amdaini, 2021-11-23:
Thank you for your prompt answer!

In fact, it's marked as record on the Treatments tab
[image]

Is it somehow possible to see it as record on the nsclient installed on parent's phone? (BTW the last nsclient version built from dev - "8.2.12-dev"  keeps crashing)

Thank you again for the amazing work you are doing for this community.

MilosKozak, 2021-11-23:
current version is 2.8.2.13-dev or beta7
the fact it has been marked as "Record only" is not visible in NS or NSClient because record in DB is the same (only not sent to the pump)

closed 2021-11-23 by MilosKozak

ToreBj, 2021-11-27:
> current version is 2.8.2.13-dev or beta7 the fact it has been marked as "Record only" is not visible in NS or NSClient because record in DB is the same (only not sent to the pump)

Seems to me that this attribute "Record only" should be added in the DB then so it can be presented/maintained for further use and reports??