nightscout/AndroidAPS #983: [3.0-beta6] Initiating bolus when connection counter is already running causes time-out

regiorunner, 2021-11-24:
Issue happened using the Dash pod.
Sometimes the bluetooth connection timer is already running when issuing a bolus command.
When that happens, the system cannot connect any more to the Dash pod and you have to wait until time-out before the system responds again. So it looks neither the origingal task nor the bolus is executed until after the time-out and a new retry.
I have attached the relevant log files.
The time the problem happens is 13:49:32:610
[AndroidAPS._2021-11-20_00-00-04_.1.zip](https://github.com/nightscout/AndroidAPS/files/7594855/AndroidAPS._2021-11-20_00-00-04_.1.zip)
.

avereha, 2021-11-24:
I don't understand what is the bug here.
After reading the logs, it looks like a bolus command timed out on connect. 
Connections can fail from time to time, if this happens often mark your phone model as 'not supported'.

regiorunner, 2021-11-25:
It is indeed no serieus problem. It happens only very occasionally. What puzzles me in the persistant impossibility to connect. Usually when I see a connection problem bringing the phone close to the Dash pod solves the problem, but when this happens it really refuses to connect and runs to time-out.
It does not seem phone related, I experienced this behavior with three different phones, roughly once per one or two Dash pods.
So, in practice not a problem, but puzzling.

avereha, 2021-11-25:
Yes, I also don't know why this happens sometimes.

avereha, 2021-12-18:
I don't think this is a bug.
The fact that you initiate a bolus where there is a pending connection it doesn't make the connection to timeout.
It would timeout anyway.

regiorunner, 2021-12-20:
As said, it is not a problem in practice. But maybe good to consider to have a message in the final version when this happens, else the user does not understand why it is not connecting. Or will the connection counter remain visible, also in the non-developer versions? (I personally like this additional info to understand what is happening).

avereha, 2021-12-29:
Yes, the connection counter will remain visible.

closed 2021-12-30 by MilosKozak