nightscout/AndroidAPS #985: remove 'do not bolus, record only' option from the treatment dialogue

avereha, 2021-11-24:
https://github.com/nightscout/AndroidAPS/issues/976 is the reason I'm removing this option from the 'Treaments' dialogue. It will still be available in the 'Insulin' dialogue.
I'm opening this issue as a reminder to also remove the code used by that option.
For now, I just hid it to see if there are some good reasons to keep it.

robertrub, 2021-11-25:
For me, it seems ok to remove that possibility from Treatments tab. As long as it is available in Insulin dialogue box, it is ok.

I never enter stuff via the treatments tab, only use it to look up what was done in the near past.

avereha, 2021-11-25:
Yes, this option will remain in the `Insulin` dialogue.

closed 2021-11-27 by MilosKozak