nightscout/AndroidAPS #987: [3.0-beta6] Eros: Message bolus stopped not cleared

regiorunner, 2021-11-25:
Only a minor detail.
When I initiated a bolus I pressed by accident the stop button, even before the bolus injection was initiated.
The bolus was initiated but almost immediately stopped again with a message. This behavior is fine of course.
That message (Bolus interrupted or stopped) remained on the screen and could not be closed.
Only by selecting another app and going back to APS cleared the message.

MilosKozak, 2021-11-25:
screenshot?

regiorunner, 2021-11-25:
Unfortunately I have no screenshot. Will try to reproduce tomorrow, but use now beta7, error occurred with beta6.

regiorunner, 2021-11-27:
In the mean time I am running beta7.
The problem is still present.
There is no difference whether you press stop button before the injection starts or after the injection started.
The overlay screen remains visible and does not disappear. I find no other way to get rid of the overlay screen than deselecting the app via the square button at the bottom and selecting APS again. Then the screen is rebuild and the overlay message disappears.

This time I have a screenshot, one when pressing the stop button before the injections started, the other when the injection already was running. The result is the same.

I hope it ismore clear now.

[image]
[image]

StephenBrown2, 2021-11-30:
Can confirm this happened to me as well, on 3.0-beta7 with Eros Pod

kristinsmotz, 2021-12-03:
Had this yesterday with beta9 on Eros pod. Originally thought it stuck because it had trouble stopping the bolus, so I ran after my daughter waving her rileylink at the pod (she left cake half eaten to go do something, which is why I was stopping the bolus after about 2 thirds was delivered).
As soon as I noticed it still didn't go away I started screen recording (remembered seeing this issue report and recognized it). After about 30 seconds I swiped away the app and reopened it and the message was gone. 
Exported logs as well. 
From the calc in Treatments tab I can see that it originally was supposed to bolus 8.65U, But the bolused amount was changed to 5.65U. So it did stop the bolus as soon as I pressed stop. 

https://user-images.githubusercontent.com/71758682/144573536-781ab104-6520-4014-8d87-ac1fe8a71107.mp4
[AndroidAPS_LOG_1638462933925.log.zip](https://github.com/nightscout/AndroidAPS/files/7648013/AndroidAPS_LOG_1638462933925.log.zip)

[image]

[image]

MilosKozak, 2022-06-21:
hopefully fixed by https://github.com/nightscout/AndroidAPS/commit/bbcbf3f90535c719fb70bbcb41c486cc7bf475ad