nightscout/AndroidAPS #988: Various possibly related alerting issues.

PedanticAvenger, 2021-11-25:
Multiple Occurrences at various times
3.0b7 (32857eb77-2021.11.23-12:12)
Oneplus 6, Android 11

Summary of Case 1:
Alert volume is erratic, typically very quiet and can't seem to be adjusted.  Sometimes the alert will start playing loudly and a couple seconds later the volume drops.  Generally noticed with cases of "pump unreachable" or "Unable to set basal" alerts.  In 2.8.2 volume was loud consistently.

Summary of Case 2:
Alerts will fire and begin to play the appropriate sound, but app doesn't always display the banner to allow for dismissal of the alert.  In 2 cases the banner appeared after ~30 seconds of waiting and in all other cases the app required a restart to stop the alarm

Summary of Case 3:
On medtronic 722 pump case, when a larger bolus is delivered, while the bolus is being delivered AAPS may try to modify basal rates and errors out trying to update basal rate while bolus is being delivered.

PedanticAvenger, 2021-11-25:
More details.
Local alerts are not configured to use system notifications (as that doesn't seem to work as a regular user of an android phone would expect).  
The "increase volume" function also doesn't seem to work. When I let the alert run for an extended period the volume remains the same.

MilosKozak, 2021-11-25:
to be able help i need logs

PedanticAvenger, 2021-11-27:
Log 1, most recent.
Example of Case 3:
[image]
[AndroidAPS_LOG_1637977009742.log.zip](https://github.com/nightscout/AndroidAPS/files/7611088/AndroidAPS_LOG_1637977009742.log.zip)

PedanticAvenger, 2021-11-27:
Earlier example, between 3:20 and 3:25 local time
[AndroidAPS_LOG_1637965259496.log.zip](https://github.com/nightscout/AndroidAPS/files/7611091/AndroidAPS_LOG_1637965259496.log.zip)
.

PedanticAvenger, 2021-11-27:
Another alert related oddity.  Carbs requested banner could not be dismissed until app was restarted.  Button shows a press but banner remains.
[image]

andyrozman, 2021-11-27:
Case 3 was previously handled with busy flag set on driver. I am not sure if AAPS is no longer considering that flag, when new commands are being queued? 
When bolus is placed, we calculate time how long bolus will need, and for that time isBusy is returning true. 

@MilosKozak is isBusy still evaluated when processing the pump queue?

MilosKozak, 2021-11-29:
pump status refresh is not requested when pump is busy

closed 2023-08-17 by MilosKozak