nightscout/AndroidAPS #989: (3.0Beta7)  NSClient suggestion for improvement

elgrupo123, 2021-11-25:
NSClient suggestion for improvement:

- Quick Wizard synchronization.  Would be very helpful for us parents.
- Possibility to disconnect or pause the pump.

We love the new NSClient, really good!!!

MilosKozak, 2021-11-26:
QuickWIzard is good idea
but NSClient will not have commands affecting direct insulin delivery because of safety

elgrupo123, 2021-11-27:
OK :(  Theoretically, the option can then be removed from the NSClient as it has no function.

elgrupo123, 2021-11-27:
please do not get this wrong. 
the idea was: we as parents could decide at any time via NS client via quickwizzard how much insulin the child should deliver at the push of a button (Quickwizzard).  Don't we enter the insulin ourselves!

kristinsmotz, 2021-12-03:
@elgrupo123 I understand your suggestion to be that the QuickWizard settings be synced between NSClient and AAPS, so the parent can adjust the settings on their phone, and those settings then are active on kid's phone as well. So if parent in NSClient changes the Quickwizard "Spaghetti" from 25 carbs to 30 carbs, and changes what boxes should be ticked for the calculation, the child when pressing their QuickWizard "Spaghetti" on the AAPS phone, the new values are chosen for that calculation. So parent would not be using the QuickWizard from NSClient phone, only configuring it for the child. 

Am I correct? 

For myself, this would make it possible to use Quickwizard, possibly instead of post-it notes in the lunchbox giving today's carb counts. If the settings could also allow setting carb time x minutes in the future with an alert after time is up, that would make breakfast prebolusing even simpler. Not to mention if we could have several quickwizards in their own tab, it would make it easier for kids to dose independently without having to remember carb counts for common meals, or checking the food tab to find carb count and entering manually in bolus calculator.

elgrupo123, 2021-12-03:
@kristinsmotz  
I think the same way too! Sponat react to wishes for children. The child presses the Qucik Wizzard sent by parents, that's it. That would be a great improvement for us parents of toddlers!

szantos, 2021-12-21:
> QuickWIzard is good idea but NSClient will not have commands affecting direct insulin delivery because of safety

For disconnect I agree.

Pausing stops insulin delivery, it's safety concern is "only" high BG.

Automation alone does no direct delivery. Therefore tuning it via NSClient would also do no direct delivery.

olorinmaia, 2026-09-23:
This is now possible in dev and will be in the AAPS 4.0.0 release. Closing issue.

Checkout running mode and the new master/client sync

closed 2026-09-23 by olorinmaia