nightscout/AndroidAPS #994: [3.0-beta7] Warning message to change pump driver after renewing Eros pod

regiorunner, 2021-11-27:
The problem occurred after changing an Eros pod.
Some time after the change a red warning bar appeared which stated: "Data is from another pump. Change pump driver to reset the pump status". The time, mentioned in the message is the time of the pod change, though the message appreared (much) later.

Below a screen shot of the message.

It looks the pump data is not really cleared or updated after de-activating the pod.

[image]

regiorunner, 2021-11-27:
There could be a relation with #973.

MilosKozak, 2021-11-29:
logs?

regiorunner, 2021-11-30:
Attached the logs of the relevant period. The pump was replaced 27 November around 08:00 hrs, the message appeared (much) later.

[AndroidAPS._2021-11-27_00-00-00_.1.zip](https://github.com/nightscout/AndroidAPS/files/7626157/AndroidAPS._2021-11-27_00-00-00_.1.zip)
[AndroidAPS._2021-11-27_00-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/7626158/AndroidAPS._2021-11-27_00-00-00_.0.zip)

MilosKozak, 2021-12-06:
https://github.com/nightscout/AndroidAPS/commit/952278a93a13530203f5fe689e61ec2d0d07cd07

closed 2021-12-06 by MilosKozak