nightscout/AndroidAPS #995: [3.0-beta7] MDT: Bolus Windows remains open

jroth-haj, 2021-11-27:
Sometimes the Bolus windows remains open.
* Bolus
* see the message "waiting for disconnect"
* see the messaage "sheduled status refresh"
* see the message "waiting for disconnect"
* the  Bolus Window remains open until I press the stop butoon.

Log send to @andyrozman 

Messages:
[image]
[image]

rustymonkey, 2021-12-01:
I'm able to replicate this as well (it seems to always occur in my test environment though I only tested while temp basals were active). Note that:
- The dialog continues to display 'Going to bolus x U' even after the bolus has been given (and the bolus was always given in my testing--it shows in both the pump and the treatments tab)
- The dialog continues to display indefinitely until the Android screen timeout kicks in. Upon re-activation of the display, the dialog is no longer displayed.
So it looks like it's mainly a UI issue, but a serious one since it confuses users as to whether the bolus was given or not, and since it presents a STOP button that actually does nothing (it can be clicked long after the bolus has completed, giving users the impression that a bolus has been stopped even when it's already been given).

A secondary issue (I'm not sure if it would be considered a separate bug) is that the bolus status dialog always disappears as soon as the screen timeout kicks in--even when the bolus hasn't yet completed.

MilosKozak, 2021-12-14:
https://github.com/nightscout/AndroidAPS/commit/201a04bfbf1c8e8f742931371ed23c895b800cbc#diff-d6f6ae060b7ba29e25a363d3dec215fe535f07682232a757f9209348f75ac039L642

closed 2021-12-14 by MilosKozak