nightscout/AndroidAPS #996: [3.0-beta7] NS Client doesn't sync extended boluses

vmichalec, 2021-11-28:
Version `3.0-beta7` - booth aaps and nsclient, nightscout: `14.2.4`, Dana R pump

I noticed that at least the extended boluses tab of the treatments is empty on nsclient side. Check images bellow - left side is AAPS, right side is nsclient. The difference can be seen also on main graph of nsclient. And probably also `IOB` shown on nsclient is affected by this.

And another minor issue on nsclient side - I'm unable to hide the fiasp-oref insulin tab. It's not part of configuration as was in previous version.

[image] [image]

[image] [image]

[image] [image]

MilosKozak, 2021-11-29:
fixed by https://github.com/nightscout/AndroidAPS/commit/3739db489f47377021639474f79e7b06bbc57ce2

MilosKozak, 2021-11-29:
fixed by https://github.com/nightscout/AndroidAPS/commit/3ee1a3a9d806df84e0e559af40df1512586a085b

closed 2021-11-29 by MilosKozak