nightscout/AndroidAPS #1003: [3.0-beta7] Dana R stats page now empty

stefanchavdarov, 2021-11-29:
2.8 version Dana R stats page was quite useful, now its empty with 3.0-beta7 
[image]

Log line at reloading stats from pump from attached log file:
`00:10:08.692 [Thread-81396] D/PUMPCOMM: [MsgHistoryDone.handleMessage():18]: History done received`
but onward the screen ends up empty
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7621481/AndroidAPS.log)

Pump: Dana R
BG source: Dexcom One on xdrip native
on Android 11, Samsung A72

MilosKozak, 2021-11-29:
did you switch to 3.0 recently?

stefanchavdarov, 2021-11-29:
Switched to 3.0 at beta-4 and following updates ever since.. but I didnt visit pump stats page till today
I wipe db for that red triangle when new sensor start goes with xdrip g6 native, thus i assume appear empty

MilosKozak, 2021-11-29:
fixed by https://github.com/nightscout/AndroidAPS/commit/918a245ed04b8726c39d6966aa0dbb7134612177

closed 2021-11-29 by MilosKozak

stefanchavdarov, 2021-11-29:
WOW thats fast