nightscout/AndroidAPS #1005: [3.0-beta7] AAPS didnt update 'x min ago' with new BG reading being same as prior one

stefanchavdarov, 2021-11-29:
3.0-beta7 doesnt update 'x min ago' when no change in BG
Screen shows "8 min ago" at phone timestamp 00:11,. Xdrip at same time was showing 2 min ago and I've been debugging why its reading wasnt delivered to AAPS. It seems it was actually delivered, but AAPS didnt update the "min ago" timestamp. 
Log - 10.4 (188.0 in mg/dl log) was delivered at 00:07 (log line below). 

I presume it happens when delta is 0 (+0.06) between prior and current value, but not certain enough.

[image]

Log line from recive
`00:07:50.721 [pool-15-thread-4] D/BGSOURCE: [XdripPlugin$XdripWorker.doWork():80]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=0.0, com.eveningoutpost.dexdrip.Extras.VersionInfo=2111161026 68eecdc-2021.11.16, com.eveningoutpost.dexdrip.Extras.Raw=0.0, com.eveningoutpost.dexdrip.Extras.Time=1638223667481, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=-1, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=-1, com.eveningoutpost.dexdrip.Extras.NoiseWarning=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=188.0, com.eveningoutpost.dexdrip.Extras.SourceDesc=G6 Native / G5 Native, com.eveningoutpost.dexdrip.Extras.SourceInfo=G6 Native, com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus=  }]`

stefanchavdarov, 2021-11-29:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7621617/AndroidAPS.log)
.. the log

MilosKozak, 2021-11-29:
when this happen again paste screenshot of BG tab too

MilosKozak, 2021-12-07:
closed for inactivity

closed 2021-12-07 by MilosKozak