nightscout/AndroidAPS #999: [2.8.2.13-dev] Pod failed with "Unknown Announcement" in treatments, and "set bolus" "command not sent" in pod history.

shafiro, 2021-11-29:
Galaxy S20 Plus.
Pod screamed 28.11@08:49 and had to be deactivated and replaced.
Treatments: "Unknown Announcement"
Pod history: "set bolus" "command not sen
[image]
[image]
t
[logs.zip](https://github.com/nightscout/AndroidAPS/files/7617718/logs.zip)
"

avereha, 2021-11-29:
The logs are from 29.11.
Can you please add logs from 28.11?

shafiro, 2021-12-06:
Happened again today. Logs attache
[info.nightscout.androidaps.zip](https://github.com/nightscout/AndroidAPS/files/7663323/info.nightscout.androidaps.zip)

avereha, 2021-12-06:
I can't find it in the logs. At what time did it happen?

avereha, 2021-12-29:
Suggesting to close this unless we find logs that contain the error.

closed 2021-12-29 by shafiro