nightscout/AndroidAPS #1012: Feature request: Make info from RED warning triangle more user-friendly

vanelsberg, 2021-11-30:
**AndroidAPS** 2.8.2.14-dev
**Build**: n/a
**Flavor**: n/a

**Feature request** (should have, next version):
When getting the RED warning icon in overview notifying duplicate BG entries in NS, user is tasked to remove the problematic entries from NS. This can be done by removing BG records from AAPS/NSClient overview. 

By clicking the RED triangle some "cryptic" info is given on the BG entries causing trouble.
Would be nice if this info could be more user-friendly:
1. Show real datetime's + BG value (in stead of timestamps with additional technical info)
2. Show the BG source for the record (is that feasible - I think NS has this info)
3. Make the NSClient view reposition to the current position after clicking "remove".
On the NSClient tab, clicking the "remove" link removes the BG entry, but then repositions at the start of the list.
For every record to be removed, user has to scroll down multiple pages to the appropriate location.

MilosKozak, 2021-11-30:
improved by https://github.com/nightscout/AndroidAPS/commit/6ba6cd6eddbeb8d0222aaf72cc5b456475395413

closed 2021-11-30 by MilosKozak

vanelsberg, 2021-11-30:
@MilosKozak Missed this one. Tnx!
You are implementing features faster then I can think of ;-)