nightscout/AndroidAPS #1024: Loop not executing every 5 mins even when a TB needs changing

AureliaJohanna, 2021-12-02:
Hi,

I've been experiencing an issue with my daughters AAPS where the loop is not executing every 5 mins even though a TB needs changing. This is obviously a safety issue if I cannot resolve.

If I manually press 'run now' in the loop tab, then loop runs. Or if I set a new action, such as a TT, then loop will run. It is connecting to the pump just fine when it runs. BGs are being pulled from BYOD app fine.

I've also noticed that Autsens seems to be changing value with each new BG. For example, I saw it change from 120 to 100 to 109 to 115 to 116 consecutively 

Actions taken:

-restared phone multiple times
-cleared AAPS cache
-swapped out battery on pump and changed battery cap
-double checked battery saving is off

I've attached some log files and a screenshot. Please let me know what other info you need. It has been happening for two days now.

Thanks,

Steve

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Andr
[AndroidAPS._2021-12-02_08-42-56_9.log](https://github.com/nightscout/AndroidAPS/files/7643298/AndroidAPS._2021-12-02_08-42-56_9.log)
[AndroidAPS._2021-12-02_08-42-56_10.log](https://github.com/nightscout/AndroidAPS/files/7643303/AndroidAPS._2021-12-02_08-42-56_10.log)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7643304/AndroidAPS.log)
oid/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/ne
[image]
w

AureliaJohanna, 2021-12-02:
Clearing cache and restarting phone seems to fix temporarily. Looking at the logs, there seems to be a profile switch event spamming but I could be wrong and have no idea where that's coming from

MilosKozak, 2021-12-02:
same as https://github.com/nightscout/AndroidAPS/issues/1021

AureliaJohanna, 2021-12-02:
> same as #1021

my phone storage is not full though? I've only used 40% of its storage capacity

AureliaJohanna, 2021-12-02:
@MilosKozak i spoke to the person that raised the other issue you closed and her phone storage was only 30% full so I’m confused as to how that’s the underlying issue and how I therefore resolve it?

MilosKozak, 2021-12-02:
what time in logs it was not executed?

MilosKozak, 2021-12-02:
sorry, you are right. i didn't realize it's 2.8 version and compared to 3.0
but the reason is not visible from logs. i see this loop invocation
```
15:23:40.846 [RxCachedThreadScheduler-4] D/APS: [LoopPlugin.invoke():352]: invoke from EventTempTargetChange
15:23:41.268 [RxCachedThreadScheduler-4] D/APS: [LoopPlugin.invoke():583]: invoke end
15:31:48.348 [RxCachedThreadScheduler-4] D/APS: [LoopPlugin.invoke():352]: invoke from EventTempTargetChange
15:31:48.757 [RxCachedThreadScheduler-4] D/APS: [LoopPlugin.invoke():583]: invoke end
15:37:24.490 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638459425000, date=02/12/2021 15:37, value=126.0, direction=Flat, raw=0.0}
15:37:24.942 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():583]: invoke end
15:42:16.799 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638459725000, date=02/12/2021 15:42, value=123.0, direction=Flat, raw=0.0}
15:42:17.242 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():583]: invoke end
15:47:17.938 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638460025000, date=02/12/2021 15:47, value=119.0, direction=Flat, raw=0.0}
15:47:18.357 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():583]: invoke end
15:52:33.456 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638460325000, date=02/12/2021 15:52, value=115.0, direction=Flat, raw=0.0}
15:52:33.861 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():583]: invoke end
15:57:25.012 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638460625000, date=02/12/2021 15:57, value=114.0, direction=Flat, raw=0.0}
15:57:25.404 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():583]: invoke end
16:02:18.145 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638460925000, date=02/12/2021 16:02, value=115.0, direction=Flat, raw=0.0}
16:02:18.564 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():583]: invoke end
16:07:18.074 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638461225000, date=02/12/2021 16:07, value=115.0, direction=Flat, raw=0.0}
16:07:18.481 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():583]: invoke end
16:12:19.442 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638461525000, date=02/12/2021 16:12, value=112.0, direction=Flat, raw=0.0}
16:12:19.857 [RxCachedThreadScheduler-7] D/APS: [LoopPlugin.invoke():583]: invoke end
16:19:08.864 [RxCachedThreadScheduler-4] D/APS: [LoopPlugin.invoke():352]: invoke from EventTempTargetChange
```

AureliaJohanna, 2021-12-02:
@MilosKozak  I believe it failed run at 20.52 in this log (so I ran a TT to get bit to run) but its been happening for two days for me

[AndroidAPS._2021-12-02_16-24-17_.log](https://github.com/nightscout/AndroidAPS/files/7645121/AndroidAPS._2021-12-02_16-24-17_.log)

AureliaJohanna, 2021-12-02:
@MilosKozak I also noticed autosens was behaving in usual way. Changing every new BG but I didn't note the exact time when that happened unfortunately

MilosKozak, 2021-12-02:
no reason visible in new log as well, sorry
you may try to turn on AUTOSENS logging and provide new logs

AureliaJohanna, 2021-12-02:
@MilosKozak okay I have turned on AUTOSENS logging. Can you confirm that you'd like me to upload another log that includes autosens when I see another missed BG?

MilosKozak, 2021-12-02:
yes

MilosKozak, 2021-12-02:
autosens logging is generating huge amount of data. maybe there will be needed upload more files
turn it off afterwards

AureliaJohanna, 2021-12-03:
@MilosKozak sods law it stopped happening consistently once I turned on autosens log last night!!! However, I think I had two missed loop runs at 11.07 and 11.32 today. Please note that I wasn't with mu daughter at the time so couldn't confirm on Loop Tab of her AAPS phone. I was monitoring via NS Client and NS.

I've added the logs.

[AndroidAPS._2021-12-03_00-00-11_68.log](https://github.com/nightscout/AndroidAPS/files/7650576/AndroidAPS._2021-12-03_00-00-11_68.log)

[AndroidAPS._2021-12-03_00-00-11_64.log](https://github.com/nightscout/AndroidAPS/files/7650574/AndroidAPS._2021-12-03_00-00-11_64.log)

AureliaJohanna, 2021-12-03:
It may not be related but I also noticed the pump connect time in NS Client said 18m ago at 11.36 and 9m ago at 11.37. I will attach screensho
[image]
[image]
ts

AureliaJohanna, 2021-12-03:
Adding additional logs around that time in case you need/want them
[AndroidAPS._2021-12-03_00-00-11_65.log](https://github.com/nightscout/AndroidAPS/files/7650650/AndroidAPS._2021-12-03_00-00-11_65.log)
[AndroidAPS._2021-12-03_00-00-11_66.log](https://github.com/nightscout/AndroidAPS/files/7650652/AndroidAPS._2021-12-03_00-00-11_66.log)
[AndroidAPS._2021-12-03_00-00-11_67.log](https://github.com/nightscout/AndroidAPS/files/7650653/AndroidAPS._2021-12-03_00-00-11_67.log)
[AndroidAPS._2021-12-03_00-00-11_69.log](https://github.com/nightscout/AndroidAPS/files/7650654/AndroidAPS._2021-12-03_00-00-11_69.log)
em

savek-cc, 2021-12-03:
Looking at the hh:mm entries in the log:
11:07
11:08 +1
11:09 +1
11:10 +1
11:11 +1
11:12 +1
11:13 +1
11:14 +1
11:15 +1
11:16 +1
11:17 +1
11:18 +1
11:19 +1
11:20 +1 
11:21 +1
11:22 +1
11:32 +10 !!!
11:33 +1
11:34 +1
11:35 +1
11:36 +1
11:37 +1
11:38 +1
11:39 +1
11:40 +1
11:41 +1
11:42 +1
so there is a gap of a full 10 minutes of nothing in those logs where absolutely nothing (not a single log entry) happened - even tough you say battery saving is disabled - there still is a 10 minute gap there; what kind of other apps are running on the device?

AureliaJohanna, 2021-12-03:
Hey

The phone is used exclusively for AAPS so the only apps I personally
initialise are the BYOD app and AndroidAPS.

Chrome will sometimes be running in the background.

I once installed an app called OpenSignal to test her schools WiFi but I
don’t open that app now.

Did the other user who had similar issue  (#1021) have a similar gap in her
logs?

Thanks

Steve

savek-cc, 2021-12-03:
Just combining all the logs, I don't see a missing loop-plugin invocation at the times mentioned? (combined "new bg event" and "loop invocation")

```
11:07:22.001 [IntentService[DataService]] D/DATABASE: [DatabaseHelper.createIfNotExists():396]: BG: New record from: DexcomG6 BgReading{date=1638529627000, date=03/12/2021 11:07, value=151.0, direction=Flat, raw=0.0}
11:07:25.060 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638529627000, date=03/12/2021 11:07, value=151.0, direction=Flat, raw=0.0}
11:07:31.964 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end

11:12:23.081 [IntentService[DataService]] D/DATABASE: [DatabaseHelper.createIfNotExists():396]: BG: New record from: DexcomG6 BgReading{date=1638529927000, date=03/12/2021 11:12, value=154.0, direction=FortyFiveUp, raw=0.0}
11:12:25.666 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638529927000, date=03/12/2021 11:12, value=154.0, direction=FortyFiveUp, raw=0.0}
11:12:26.124 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end

11:17:19.167 [IntentService[DataService]] D/DATABASE: [DatabaseHelper.createIfNotExists():396]: BG: New record from: DexcomG6 BgReading{date=1638530227000, date=03/12/2021 11:17, value=146.0, direction=Flat, raw=0.0}
11:17:21.772 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638530227000, date=03/12/2021 11:17, value=146.0, direction=Flat, raw=0.0}
11:17:22.221 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end

11:22:22.936 [IntentService[DataService]] D/DATABASE: [DatabaseHelper.createIfNotExists():396]: BG: New record from: DexcomG6 BgReading{date=1638530527000, date=03/12/2021 11:22, value=142.0, direction=Flat, raw=0.0}
11:22:25.531 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638530527000, date=03/12/2021 11:22, value=142.0, direction=Flat, raw=0.0}
11:22:26.000 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end

11:27:20.497 [IntentService[DataService]] D/DATABASE: [DatabaseHelper.createIfNotExists():396]: BG: New record from: DexcomG6 BgReading{date=1638530827000, date=03/12/2021 11:27, value=150.0, direction=Flat, raw=0.0}
11:27:23.109 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638530827000, date=03/12/2021 11:27, value=150.0, direction=Flat, raw=0.0}
11:27:23.575 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end

11:32:25.617 [IntentService[DataService]] D/DATABASE: [DatabaseHelper.createIfNotExists():396]: BG: New record from: DexcomG6 BgReading{date=1638531127000, date=03/12/2021 11:32, value=153.0, direction=Flat, raw=0.0}
11:32:28.197 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638531127000, date=03/12/2021 11:32, value=153.0, direction=Flat, raw=0.0}
11:32:35.135 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end

11:37:21.699 [IntentService[DataService]] D/DATABASE: [DatabaseHelper.createIfNotExists():396]: BG: New record from: DexcomG6 BgReading{date=1638531427000, date=03/12/2021 11:37, value=147.0, direction=Flat, raw=0.0}
11:37:24.272 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638531427000, date=03/12/2021 11:37, value=147.0, direction=Flat, raw=0.0}
11:37:24.712 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end

11:42:23.048 [IntentService[DataService]] D/DATABASE: [DatabaseHelper.createIfNotExists():396]: BG: New record from: DexcomG6 BgReading{date=1638531727000, date=03/12/2021 11:42, value=138.0, direction=Flat, raw=0.0}
11:42:25.686 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638531727000, date=03/12/2021 11:42, value=138.0, direction=Flat, raw=0.0}
11:42:26.144 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end
```

But if you compare what is happening, you can see that it just "ends" more or less directly after "getMealData" when it is "not executed" - but "getMealData" takes virtually "forever" at this stage:

```
11:07:25.060 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638529627000, date=03/12/2021 11:07, value=151.0, direction=Flat, raw=0.0}
11:07:25.063 [RxCachedThreadScheduler-8] D/APS: [OpenAPSSMBPlugin.invoke():161]: invoke from AutosenseCalculation for BgReading{date=1638529627000, date=03/12/2021 11:07, value=151.0, direction=Flat, raw=0.0} tempBasalFallback: false
11:07:31.962 [RxCachedThreadScheduler-8] D/APS: [Profiler.log():13]: >>> getMealData() <<< executed in 6521 miliseconds
11:07:31.964 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end
```

compared to a complete run:

```
11:12:25.666 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():352]: invoke from AutosenseCalculation for BgReading{date=1638529927000, date=03/12/2021 11:12, value=154.0, direction=FortyFiveUp, raw=0.0}
[... cut ...]
11:12:26.118 [RxCachedThreadScheduler-8] D/APS: [APSResult.isChangeRequested():338]: DEFAULT: Closed mode
11:12:26.120 [RxCachedThreadScheduler-8] D/APS: [APSResult.isChangeRequested():338]: DEFAULT: Closed mode
11:12:26.120 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.applyTBRRequest():678]: applyAPSRequest: Rate: 0.54 U/h (180.00%) 
11:12:26.121 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.applyTBRRequest():731]: applyAPSRequest: setTempBasalAbsolute()
11:12:26.124 [RxCachedThreadScheduler-8] D/APS: [LoopPlugin.invoke():583]: invoke end

```

savek-cc, 2021-12-03:
@MilosKozak I wonder if it's throwing an exception somewhere (shortly after, as the only think from the contraints that is checked is the IOB) after https://github.com/nightscout/AndroidAPS/blob/2.8.2/app/src/main/java/info/nightscout/androidaps/plugins/aps/openAPSSMB/OpenAPSSMBPlugin.java#L204

MilosKozak, 2021-12-07:
the root cause is too much concurrent changes in DB which is preventing AS calculation to finish.
I'm unable to resolve it in 2.8
In 3.0 this has been improved a lot and there is more usefull logging in this area. Wait for 3.0 and then create new issue if the problem persists.

closed 2021-12-07 by MilosKozak

AureliaJohanna, 2021-12-07:
Okay - thanks Milos - much appreciated.

If there could be some sort of post ‘Loop Run’ check that could send a
notification if it hasn’t executed, I think that would be helpful. Not sure
if that would be possible.

Thanks again for all you do

Steve