nightscout/AndroidAPS #1027: [3.0beta9] Feature request treatments tab

kristinsmotz, 2021-12-03:
I'd like to request back a feature I used frequently in the previous version (2.8.2)
In the old Treatments tab, for each temp basal set, there were several calculations/numbers shown. 
Planned duration, ratio, actual duration, Insulin given, and current effect on IOB (in green)

[image]

In 3.0, only ratio, actual duration, and current effect on IOB is shown. 

[image]

I really miss seeing the insulin given from that temp basal. It gave me a better view of for instance, how much higher the previous meal bolus should have been to avoid the high, etc. 

Could you consider bringing back more information for each line if it is available?

MilosKozak, 2021-12-03:
this is not planned. to be honest I cannot imagine how anyone could exstract from this list anything usefull. for human brains we have graphs on overview

closed 2021-12-03 by MilosKozak

kristinsmotz, 2021-12-03:
Haha @MilosKozak I love your comments :D My brain may not be exactly human, but I liked to be able to quickly calculate how much extra insulin was given during a high. Like after breakfast bolus, if bolus is not high enough, I like to add up how much extra insulin given during the 4 hours until lunch, to get a sense of how much I would need to strenghten IC for breakfast in order to avoid her going high. Takes some of the guesswork out of it, like if IC for breakfast is now 6 (45 carbs gives 7.5U), and 2 extra U were given by temp basals before lunch and she is still high, I will know that I can skip reducing IC to 5.5 (45 carbs gives 8U) and go straight down to 5 (9U) or even 4.5 (10U). If numbers are in treatment tab I can avoid scrolling back on NS graph to see what IOB was at lunchtime and guess/calc and guesstimate how much was remaining breakfast bolus, how much was temp basals. I don't always (though quite frequently) sit and stare at my phone to note down what happens in real time, and I find scrolling NS not very helpful for this issue.