nightscout/AndroidAPS #1031: [Enhancement] Add today's TIR statistics.

bchris21, 2021-12-04:
All of us want to have the best TIR throughout the day. I personally use xDrip's extra info line though to see how am I doing today.

[image]

Is it possible to have this information in AAPS too?

Statistics have data from previous day and after but not today's. I don't know if today's TIR data can be squeezed somewhere in main screen, maybe here:

[image]

Thanks!

Alamo04, 2021-12-04:
I love that status line in XDrip, but didn't like to see that in AAPS main screen.

XDrip has not mutch information's on its main screen, AAPS a lot more.

And I don't like to scroll, to see the mutch more important information's like IOB/COB and/or sensitivity curves.

bchris21, 2021-12-04:
> I love that status line in XDrip, but didn't like to see that in AAPS main screen.
> 
> XDrip has not mutch information's on its main screen, AAPS a lot more.
> 
> And I don't like to scroll, to see the mutch more important information's like IOB/COB and/or sensitivity curves. 

Having an option to enable/disable it in main screen would be ideal (rather than making it persistent for everyone) but I don't know if it is feasible.

I understand it's a clutter for many people but may be useful for other.

Adding it only in Statistics page is also a big benefit. Main screen is just a personal recommendation.

kristinsmotz, 2021-12-07:
I would also like this, either in Overview or Statistics. If put in statistics, a line of "up to now values for today" for all the stats (TDD, bol, bas, low, in, high) would be very helpful. I currently use Xdrip for viewing TIR/high/low %, and Nightscout reports to see TDD until now.

szantos, 2021-12-21:
Many status line infos could be helpful:

TIR (lo/in/hi), avg daily SGV, A1c, Sum of daily insulin and I:C could all help to see sensitivity changes.

Total daily carbs (from 00:00 until now) could help you to know how much you can eat, if aiming at a fixed carb amount per day. If tracking correction carbs, also helpful to see sensitivity changes.

Total daily carbs is currently the only thing, I let xDrip download the treatments.

Side note:
If displaying I:C it should be considered to make the division after subtracting the 100% basal amount from the insulin (to see the average I:C for bolus).

LuisPuertollano, 2022-03-06:
Hi, I found this idea really good,....at least to have a line for the actual "day-in-progress" for TDD, TIR, etc would be really nice
[image]

Hopefully it can be done :-)
Thank you!

closed 2026-10-15 by MilosKozak