nightscout/AndroidAPS #1035: [3.0-beta9] Automation Disconnect pump entries 2/m?

chrisbev, 2021-12-05:
For me at the bottom of the Auto Tab when the loop has the 'Disconnect pump' case a large number of 'Disconnected' and 'Loop disabled' lines appear. About 2 per minute?
To aid diagnostics in future could the time be added to these lines please?
3.0-beta9, Samsung 41, FSL2 Patched LL,  xDrip
Log attached event 4:00am.
[AndroidAPS_LOG_1638679328398.log.zip](https://github.com/nightscout/AndroidAPS/files/7655485/AndroidAPS_LOG_1638679328398.log.zip)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

chrisbev, 2021-12-05:
Just looked at Auto Tab in 'Suspend loop' mode and that for me throws lots of 'Loop is disabled' lines.  If not reproducible I can send more logs/pics.

MilosKozak, 2021-12-05:
if loop is disabled automation rules are not processed and 'Loop disabled' appear in automation log every minute (1 minute is automation schedule). this is by desing

chrisbev, 2021-12-05:
OK if by design then close but for info if you have time why is the auto schedule 1 minute the rules appear to run every 5 mins and I presume when a new BG triggers AAPS to wake up?  Is it just 1 min for Disconnect or Suspend check?
It may be 'nice' to have the time added if you feel this advantageous and use this Issue as an 'enhancement request' or would you like a separate Issue? 
With Disconnect pump for 1 minutes I got 5 pairs of Disconnected and Loop is disabled not just one pair per minute but I do have 5 rules? Again hardly a problem if by design?

MilosKozak, 2021-12-06:
there are many triggers (data changes) leading to automation loop run. so in reality it can be more often than 1 min. thus this 5 min time limit