nightscout/AndroidAPS #1036: Autosense is reset (Oref1) on canceling a temp %Profile

vanelsberg, 2021-12-05:
**AndroidAPS 2.8.2.14-dev** (and earlier commits)
**Build:** d7a4740e7
**Flavor:** Fullrelease

**Problem description:**
Autosense is reset on canceling a %Profile (Oref1)

Actual the "problem" is that there is no way to *cancel* a temp profile?
Only way is setting a 100% profile "without duration". But by ORef1 definition this resets autosense.

**Additional information**

Running AAPS latest dev builds, Oref1, OpenAPS SMB, with Autosense and UAM enabled in settings.
- Autosense is reset to 100% after canceling a temp %Profile before the original duration has passed.
- As soon as I "cancel", on the SMB tab, autosense data show *ratioLimit: "(XX of 48 values)"*.
- Autosense will be activated when XX reaches 48 which can take up to 8 hours or more.

*Note: autosense it not reset when using 'Sensitivity WeightedAvarage'.*

**Repro:**
Configuration:
Sensitivity Oref1, OpenAPS SMB with autosense and UAM enabled.

1. Confirm autosense is active, showing a value other the "100%" on overview and *ratioLimit: ""*)
2. Set a profile of 110% for 10 minutes
3. When the temp profile ends, confirm autosense is still active
4. Set a profile of 110% for 120 minutes
5. After about 10 minutes "cancel" the temp profile by setting a 100% profile with no duration
6. Confirm autosense was reset: showing a value of "100%" on overview and *ratioLimit: "XX of 48 values"*)

Autosense will activate again sometime after Autosense data *ratioLimit: "(XX of 48 values")* changes to *ratioLimit: ""*
This may take 8 hours or more.

**Log files:**
[AndroidAPS_LOG_1638697480059.log.zip](https://github.com/nightscout/AndroidAPS/files/7655678/AndroidAPS_LOG_1638697480059.log.zip)

**Screenshots**
Screenshot show as described under "Repro":
- 09:43h: First 10 mins temp temp profile switch ended at the duration set, 
- 10:12h: Second 15 mins temp profile was ended after 10 minutes by setting a temp profile of 100% with no duration.

[image]
 end when duration is passed, second temp profile switch was canceled setting a 100% profile with no duration.
[image]
[image]
[image]

MilosKozak, 2021-12-05:
yes this is by definition

vanelsberg, 2021-12-05:
@MilosKozak Then how do I cancle a temp %Profile?

vanelsberg, 2021-12-05:
Then how do I cancle a temp %Profile? :-|

Do agree that changing profile, switching profile e.g. setting a temp profile without duration by definition resets autosense.
Also agree that cancelling with 100% profile with no duration is not actually a "cancel" and as such by definition also resets autosense.

So conclusion would be:
- AAPS User *can* set a temp profile that automatically ends after the set duration *without resetting autosense*
- AAPS User *can not* end/cancel a running temp profile.

@MilosKozak
**Proposing a feature request:**
Allow user to cancel a running %Profile? (without autosense resetting)
One would think that technically ending a temp profile through a timed or user event would have the same action?

If this makes any sense to you I'll post this as a feature request / enhancement?

vanelsberg, 2021-12-05:
My final conclusion, when on ORef autosense, by definition:
- Setting a profile without duration will reset autosense.
- Set a temp profile with duration will not reset autosense.

AAPS is missing the option for user to prematurely end a running temp profile.

chrisbev, 2021-12-06:
I appreciate the need to reset on pump site change as history could be way off but for profile change what is the rational? I have wondered before the resetting if say I have AS at 120% (profile at 100%) and do a switch to 120% does AS settle at 120% or 100%? (No other changes and can't test as my body is not that consistent!). I have asked this months ago but had no replies (fb and discord).  Depending on the answer then why not make the algorithm just carry on and compensate for the PS and no need to reset unless there is another need to reset that I am not aware of?  After all it is just insulin doesn't matter if it from a basal profile or a bolus?  If I am way off please disregard....

kristinsmotz, 2021-12-07:
I would also like a good option to cancel a temporary profile change. Currently using a "workaround" to keep autosens by setting a new temp profile change at 100% with a duration slightly longer than current running temp profile. But then that one is never cancelled, just keeps counting end time in negative numbers until next profile change (temp or permanent). Seems the system won't process ending a temp profile if it is identical to the last permanent profile change.

MilosKozak, 2026-10-15:
covered by scenes now

closed 2026-10-15 by MilosKozak