nightscout/AndroidAPS #1040: 2.8.2.14-dev Bolus nowhere to be found

Freloner, 2021-12-06:
**AndroidAPS 2.8.2.14-dev** 
**Build**: 8783e67a9
**Flavor**: Fullrelease

**Problem description**:
I did a manual Bolus at around (maybe between) 09:25 & 09:50, then I noticed that my pod is ticking a lot due to SMBs, I checked AAPS, found that my BG is rising fast & the bolus is nowhere to be found, no trace at all.

**Log files:**
[AndroidAPS._2021-12-06_00-00-01_.2.zip](https://github.com/nightscout/AndroidAPS/files/7660212/AndroidAPS._2021-12-06_00-00-01_.2.zip)

**Screenshots:**

[image]
[image]
[image]
[image]

avereha, 2021-12-06:
I do not find any bolus in the logs.

Freloner, 2021-12-07:
> I do not find any bolus in the logs.

Me neither..

robertrub, 2021-12-07:
Had the sane problem this morning. No carb entries'and no bolus.
[image]
[image]

robertrub, 2021-12-07:
[image]
[image]

robertrub, 2021-12-07:
It might be a _user error_ in my case as I can't find any reference to any carbs or bolus in my logs. I'll keep watching.

avereha, 2021-12-29:
Adding more logging to all the dialogues: https://github.com/nightscout/AndroidAPS/pull/1109

avereha, 2022-01-17:
Suggesting to close this issue and reopen when we have logs that show the bolus.

robertrub, 2022-01-17:
Ok. You can close it AFAIK. I never had the issue again. In my case, it was a _user error_

closed 2022-01-17 by Freloner