nightscout/AndroidAPS #1045: 3.0beta9 Unable to Build

mfranzelihs, 2021-12-07:
Hello,
trying to build 3.0 Beta 9, but I get an error every time I try. Please see screenshot
[image]
.

vanelsberg, 2021-12-07:
I have no problems building from current dev. Double check documentation on building.
Make sure you have updated Android Studio to latest version.
And maybe checkout this:
https://docs.google.com/document/d/1oc7aG0qrIMvK57unMqPEOoLt-J8UT1mxTKdTAxm8-po

MilosKozak, 2021-12-07:
you may try clear studio cache

mfranzelihs, 2021-12-07:
Got it working, thank you. That document helped. I was not downloading the right branch correctly. 

However, any idea why it says I need to rebuild in just a few days?

closed 2021-12-08 by MilosKozak