nightscout/AndroidAPS #1047: Upload BG values to Nightscout after switching from BYOD to xdrip

CaptainBalou, 2021-12-07:
I was previously working with AndroidAPS dev branch and BYOD app for BG values. Upload was done via AndroidAPS and worked fine.
Then I switches over to beta7 build to test the newest code. I have not taken a look onto the transferred BG values but locally all looked fine.
Because of testing something which was fixed in beta7 branch I switched to dev again. Because of issues with BYOD app I also switched to xdrip.
After two weeks of working with dev and regularly updating it I realized that I don't have BG values in my Nightscout instance anymore. Therefore also some gadgets I use have no values to display.

I did some tests:
- Use a slightly older dev build. Result was no bg values in Nightscout.
- Check if other uploads from AndroidAPS to Nightscout work. Result was 'yes' I have other things like SMBs and so on which are uploaded. But no BG values.
- Deactivated AndroidAPS upload to Nightscout and used xdrip to upload data. Result 'yes' all was uploaded successfully including the bg values

I'm currently unsure what to check to get this issue isolated. Can someone assist in deciding what to check? If logs are important I can turn back the configuration from xdrip to AndroidAPS and gather logs. Possible but takes time for me at the moment.

CaptainBalou, 2021-12-07:
But even if I see BG value on the top right of Nightscout the whole site looks not ok because I don't see a graph line. So something is wrong somewere. I hope someone can assist.

[image]

MilosKozak, 2021-12-08:
I'm not sure I understand it well.
AAPS doesn't upload BG data to NS when xDrip bg source is selected
AAPS can upload data for other BG sources

CaptainBalou, 2021-12-08:
Sorry,was a confusing summarization of all I saw.

I cannot answer that question precisely. All I can say right now is that:
If I use xdrip to upload data to nightscout I see BG values.
If I use AndroidAPS to upload data to nightscout I see no BG values.

CaptainBalou, 2021-12-08:
The greay dots came from xdrip which I used the past hours to upload bg values.
The bolus and SMB information came after I switched of xdrip and activated AndroidAPS upload.
Now new current bg values are uploaded anymore and I just get "---" and not grey dots anymore.

[image]

I used this setup to centrally upload everything to Nightscout and it has never had any problems. 
But I don't know when it stopped working. My fault.

vanelsberg, 2021-12-08:
What is your BG source in AAPS?

CaptainBalou, 2021-12-08:
I am using Dexcom G6.

I was using BYOD app, got the bg values in Aaps and these have been uploaded by AAPS.

Because of BYOD issues I now use xdrip and assumed it works the same. xDrip sends to AAPS and AAPS uploads it.

That doesn't work that way as I understand you?

vanelsberg, 2021-12-08:
What @MilosKozak was trying to tell you is that _when xDrip sends to AAPS and AAPS **does not** upload BG value's_.

Recommended configuration is:
- BYODA as BG source for AndroidAPS which will then be the only  application sending BG values to NS.
BYODA supports broadcasting to xDrip, so you can still use it *except* for uploading to NS (if you it will result in duplicate BG values causing current AAPS to block closed looping).

Btw: If you haven't done already, Please joint the @androidaps channel at Discord: [Invite](https://discord.gg/TbnTX4UF)
for latest updates. Lot of users there that are happy to help out...

closed 2021-12-09 by MilosKozak

CaptainBalou, 2021-12-09:
Hello @MilosKozak , hello @vanelsberg ,

oh @MilosKozak your [comment](https://github.com/nightscout/AndroidAPS/issues/1047#issuecomment-988607163) was no question?! Thanks for clarifying this @vanelsberg . I'll activate the xdrip upload as well. Thanks. I thought it is a bug.

The setup of using BYOD app was prefered by me for a very long time. But past month I decided to do an major release update of the BYOD app and since I'm using that it constantly crashes. :-( So I switched to xdrip quick and dirty to at least have bg values in AAPS. Long term I want to switch back but I've to identify the issue first.

I'm on discord too. Thanks for reminding me. But I've had not a high answer rate and it's hard to follow conversations if you're not there all day. All those wonderfull developers in the community are so busy and I always feel it is easier to handle such things in an issue ticket. I'm possibly too much relying on ITIL schemes. So I decided to do so - I should start to separate business and private. ;-)

Thanks for your time and thanks for reading all that.

Regards
Marco