nightscout/AndroidAPS #1049: I use beta 9 with a combo pump and every time I click on the pump disconnect the app crashes

mordechi-david, 2021-12-08:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new
It happens every time, I use Android 10 and my logs go up through fabric by email md0527600652@gmail.com
תודה רבה

mordechi-david, 2021-12-08:
זה קרה גם עכשיו ב 6:59

swissalpine, 2021-12-08:
Same setup (Combo und last Beta/dev): Pump disconnect works like a charm!

mordechi-david, 2021-12-08:
For me it collapses again and again here is a screenshot of a disconnected
pump and basal no 0

mordechi-david, 2021-12-08:
Attaches diaries
[AndroidAPS._2021-12-08_14-50-29_.0.zip](https://github.com/nightscout/AndroidAPS/files/7676114/AndroidAPS._2021-12-08_14-50-29_.0.zip)
[AndroidAPS._2021-12-08_14-49-26_.0.zip](https://github.com/nightscout/AndroidAPS/files/7676116/AndroidAPS._2021-12-08_14-49-26_.0.zip)

closed 2021-12-15 by mordechi-david