nightscout/AndroidAPS #1050: Synchronisation error - changes NS profile are not correct synched to AndroidAPS

peterleimbach, 2021-12-08:
Reporting bugs
--------------
- I use Beta 8 and on nightspot side 14.0.7. The error occurred at around 10:10 o'clock. Sorry I can't get this information better.
- I added
  a. log file,
  b. screenshot from ns profile editor and
  c. screenshot vom AndroidAPS
- The error occurred when I switched back from 4 IC values per day to hourly based IC values. Therefore I added in the NS profile editor further entries for every hour and saved the profile.
- On ns side the hourly added values are correct but on AndroidAPS there are new entries but the time information is missing and I had to add this manual.
- The second issue I see but I don't know where it comes from (NS or AndroidAPS) ist that the timestamp since the profile is in place (gültig in German) is not correct
[AndroidAPS_LOG_1638958717120.log.zip](https://github.com/nightscout/AndroidAPS/files/7675284/AndroidAPS_LOG_1638958717120.log.zip)
[image]
[image]

.

MilosKozak, 2021-12-08:
did you follow this while setting profile?
https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Releasenotes.html#changes

peterleimbach, 2021-12-08:
I think I violated "Record modification from NS is not allowed" when I changed the profile with Nightscout profile editor if I understand this right.

MilosKozak, 2021-12-08:
I don't know this message

peterleimbach, 2021-12-08:
That's part of the release notes. I thought I used a not already implemented functionality.

"NSClient rewritten for better synchronization and more detailed customization @MilosKozak

Record deletion from NS is not allowed (only invalidation through NSClient)
**Record modification from NS is not allowed**
Sync setting available without engineering mode (for parents)
Ability to resync data"

peterleimbach, 2021-12-08:
I think I now found what you mean in the release notes.

No, I did not clone the entry. I just changed and saved it. Somehow I have not seen this in the release notes.

closed 2021-12-09 by MilosKozak