nightscout/AndroidAPS #1051: [Enhancement] text to match the colour of the graph

tonydeleon9, 2021-12-08:
Would it be possible to change the colour of the text so that it matches the graph? so iob would be blue and dev slope would be yellow as per attached pic? thx

tonydeleon9, 2021-12-08:
https://linksharing.samsungcloud.com/97kyuAWKWftx

swissalpine, 2021-12-08:
Please keep in mind that blue letters on a dark background is not easy to read. There are many colored texts in AAPS which are problematic.

Philoul, 2021-12-21:
@tonydeleon9 can you please share you image directly here (link is expired...)

olorinmaia, 2026-10-06:
As I read this these colors have been implemented in v4. It's also hard to understand the issue when the picture link has expired. Closing issue. Feel free to open new if you don't think this is covered in v4.

closed 2026-10-06 by olorinmaia