nightscout/AndroidAPS #1054: in nsclient tab on Jelly 2 "Warteschlange" to long as word that I can't see the number of items

peterleimbach, 2021-12-08:
beta 9 app

I use a Jelly 2 phone. The display is 3" small.

**In the tab nsclient the name of the queue in german is with "Warteschlange" to long to read the number of elements in the queue.**

To fix this I changed
<string name="queue">Warteschlange:</string>
to
<string name="queue">Queue:</string>
in the strings.xml for the german language.

As the phone is very, very good and some more are using it I suggest to change in the standard.
A more german translation could be "Schlange" but I think the englisch vocabular is common in german to.

swissalpine, 2021-12-09:
There are numerous translations that are well-intentioned and accurate, but did not have the UI in mind when translating, at least not that of very small screens.
Since crowd-translation constantly revises the translations, there is no sense in changing it, it is - well-intentioned - overwritten again shortly after.
In my opinion, the best workaround is to delete all German translation files - then it almost always fits on the Jelly 2.
Or, as an "enhancement", a setting that ONLY the original English text is displayed.

MilosKozak, 2021-12-09:
you can offer different translation on Crowdin but ..... it will never be perfect on very small screens.
as a workaround you can rotate screen to landscape

closed 2021-12-09 by MilosKozak